dresden-elektronik/deconz-rest-plugin #8: Update This Repository!

SpencerSharkey, 2015-08-13:
Since deCONZ 2.0.0, this repository is out of date.

Building this plugin and applying it to 2.0 breaks the web-app functionality as it's old.

When can we expect to see this repository updated with the latest rest plugin source?

Thanks

SpencerSharkey, 2015-12-16:
Around the time of August 12th, Dresden claimed they were going to update this repository within a week or so.

It never happened. I emailed back a month later, inquiring once more about the possible updated source code. They ensured it'd be uploaded by the end of the week.

We can't fix problems with the current REST API without the latest source code. You guys bundled a binary plugin with the latest release but never supplied the code.

Very frustrating, Dresden-Elektronik. I sent another inquiry a week or so ago and now they have resorted to ignoring my email.

How hard is it to git commit/push this repository?! GAH!

SpencerSharkey, 2015-12-16:
Oh, and another thing:

Since deCONZ 2.x, ZigBee Green Power device indications are received properly. Hue Taps now work, yay! But, there are still no officially updated development headers [deconz-dev](https://www.dresden-elektronik.de/rpi/deconz-dev/) 

This is lame. I was able to hack together some of my own  headers for the green power related methods. For anyone interested in hooking into GreenPower/Tap commands, here:

```c++
// GreenPower.h
#include <QObject>
#include <stdint.h>

namespace deCONZ
{
    
    class GpDataIndicationPrivate;
    /*! Address modes used to specify source and destination addresses. */
    enum ZGPCommandId
    {
        E_GP_IDENTIFY = 0x00,
        E_GP_OFF = 0x20,
        E_GP_ON,
        E_GP_TOGGLE,
        E_GP_LEVEL_CONTROL_STOP = 0x34,
        E_GP_MOVE_UP_WITH_ON_OFF,
        E_GP_MOVE_DOWN_WITH_ON_OFF,
        E_GP_STEP_UP_WITH_ON_OFF,
        E_GP_STEP_DOWN_WITH_ON_OFF,
        E_GP_COMMISSIONING = 0xE0,
        E_GP_DECOMMISSIONING,
        E_GP_SUCCESS,
        E_GP_CHANNEL_REQUEST,
        E_GP_COMMISSIONING_REPLY = 0xF0,
        E_GP_CHANNEL_CONFIGURATION = 0xF3
    };
    
    class DECONZ_DLLSPEC GpDataIndication
    {
    public:
        /*! Constructor. */
        GpDataIndication();
        /*! Copy constructor. */
        GpDataIndication(const GpDataIndication &other);
        /*! Copy constructor. */
        GpDataIndication &operator=(const GpDataIndication &other);
        /*! Deconstructor. */
        ~GpDataIndication();
        
        uint8_t gpdSrcId() const;
        uint8_t frameCounter() const;
        ZGPCommandId gpdCommandId() const;
        QByteArray &payload() const;
        
        
        /*! Reads a ZigBee standard conform indication from stream. */
        void readFromStream(QDataStream &stream);
    
    private:
        GpDataIndicationPrivate *d_ptr;
        Q_DECLARE_PRIVATE(GpDataIndication)
    };
    
    
    class DECONZ_DLLSPEC GreenPowerController: public QObject
    {
        Q_OBJECT
    
    public:
        /*! Constructor. */
        GreenPowerController(QObject *parent);
        /*! Deconstructor. */
        virtual ~GreenPowerController();
        /*! Get the singleton instance of the GreenPowerController. */
        static GreenPowerController *instance();
        
        void processIncomingData(const QByteArray &);
        
    
    Q_SIGNALS:
        void gpDataIndication(const GpDataIndication &);
    };
}
```

To connect the indications to your plugin, add it to your slots in your plugin's header file, create a pointer to the deCONZ::GreenPowerController as well.

```c++
public Q_SLOTS:
    void gpDataIndication(const deCONZ::GpDataIndication &ind);
private:
    void disconnectSockets();
    deCONZ::GreenPowerController *GPC;
```

In your plugin constructor, initialize a GreenPower controller, and connect the signal from it to your app's GpDataIndication like so:
```c++
/*! Plugin constructor */
    GPC = deCONZ::GreenPowerController::instance();
    DBG_Assert(GPC != 0);

    connect(GPC, SIGNAL(gpDataIndication(const deCONZ::GpDataIndication&)),
            this, SLOT(gpDataIndication(const deCONZ::GpDataIndication&)));
``` 

And finally, here's an example of what you can do with the gpDataIndication.
```c++
void Fusebox::gpDataIndication(const deCONZ::GpDataIndication &gp)
{
    QVariantMap json;
    json.insert("layer", QVariant(QString("aps")));
    json.insert("type", QVariant("greenpower"));

    json.insert("srcAddress", QVariant(gp.gpdSrcId()));
    json.insert("commandId", QVariant(gp.gpdCommandId()));

    json.insert("payload", QVariant(gp.payload().toBase64()));

    DBG_Printf(DBG_INFO, "->[GP Indication] Source: %u Command: %u Payload: %s\n", gp.gpdSrcId(), gp.gpdCommandId(), qPrintable(gp.payload().toBase64()));

    QString errMsg;
    QVariant jsonVariant = QVariant::fromValue(json);
    socketSend(QJson::encode(jsonVariant, &errMsg, 0).toUtf8());
}
```

Hector47, 2015-12-16:
I email them on 20th of November about this issue and some other stuff. Their reply was that the GitHub will be updated automatically when a new release will be available and that their next release will be in the end of December.

I think we may need to find/create a wiki and/or a forum to share our code, issues, ideas, showcase,...

jwilker2, 2015-12-16:
Very very interested in seeing updates as well!

SpencerSharkey, 2015-12-25:
It looks like we may be missing our deCONZ gift this Christmas!

SpencerSharkey, 2016-01-22:
Sigh...

xfbs, 2016-01-27:
Hey, what is the current status of the REST API, does it support the latest version of deCONZ?

SpencerSharkey, 2016-02-05:
No, it doesn't. We've been waiting for **months** for an update from Dresden...

r3pek, 2016-03-16:
Maybe as an Easter present? maybe not.... :-/

SpencerSharkey, 2016-03-17:
They continue to add features to their application with new releases, but fail to update this code. I've given up.

LunaSela, 2016-03-17:
I have given up hope too. I regret I've bought the RaspBee mostly because of the open source REST plugin. I even feel  beeing  cheated. 

Theoretically, it would be possible to develop an own REST plugin, but it looks like the development headers will have the same fate.

As for now I will have to disrecommend anybody who asks me to buy RaspBee shields.

willem4ever, 2016-03-17:
Yeah, really a shame, I suppose it is no longer their interest now that they have ZigBee Light Link certified products ...

r3pek, 2016-03-17:
@LunaSela 
I'm actually thinking on using RaspBee for HA. Are there any other alternatives??

xfbs, 2016-03-17:
@r3pek I have a Raspberry Pi with the RaspBee premium gateway and that works quite well. The REST API is functional, you just don't get any (up-to-date) source code to go along with it, and also the headers are missing so you can't write C++ extensions. For my purposes it works fine tho, just hack a small Ruby script together or use `httpie` on the CLI to controll stuff :smiley:

LunaSela, 2016-03-17:
No open source ones I tested.

There are some other projects like

[http://dog-gateway.github.io/](url)

But I have not how well they work..

At the moment I reverted to Hue, which has a reasonable REST API too. (which looks very very similar, as the deCONZ REST API seems to be a clone of the Hue). The UPnP part of deCONZ does not behave well. 

I will wait some more time before I buy any new hardware.

Shame upon Dresden Elektroinik. I would have added support for deCONZ in my application in development. Now I have chosen Hue as my primary platform.

I am doing ZLL (light bulbs, switches) only, so I cannot say anything about the other HA stuff

r3pek, 2016-03-17:
@xfbs 
as long as deCONZ works, controlling everything, i'm fine with that. Still the rest API would be nice so that everything can be controlled by a phone.

@LunaSela
dog-gateway looks nice but it's for zwave, not zigbee. Well, right now, i really don't know what's best :)

SpencerSharkey, 2016-03-17:
We could develop our own REST plugin, but at what expense? I was able to write a plugin that wraps most of the C++ API calls into a nice TCP Socket API... but even deCONZ is pretty unreliable as a base application. The benefit was I could quickly add functionality in an external application, say written in NodeJS or Python, that opens a REST API.

I've created a Slack team (web-based chatroom, has desktop/mobile apps, great fun!) that we can all participate in to share ideas, solve problems, and meet new people. Join it at https://rpizigbee.herokuapp.com/

P.S. Has anyone gotten a Samsung SmartThings Multi-purpose sensor working with deCONZ?

Hector47, 2016-04-12:
A new branch named "V2_03" is available :heart_eyes:.
The commit's comment seem to say that it will be easier for them to push updates to this repository.

manup, 2016-10-25:
Since push source code is now synced at latest with public releases the issue will be closed.
There is still room for improvements on this repository and documentation whic will be addressed in future releases.

Regards

closed 2016-10-25 by manup