dresden-elektronik/deconz-rest-plugin #9: deCONZ REST headless

remiphilippe, 2016-06-02:
Hello,
I understand there is no headless version of deCONZ today. Any plans for that?

Thanks

manup, 2016-06-02:
Hi Remi,
A headless version is requested quite often. Currently we don't have an ETA for this feature but there is already progress on this matter for some time to separate the core from the GUI.

Regards

ertanden, 2016-12-28:
Waiting for this... I hope it doesn't take much longer.

manup, 2017-04-12:
It's not the best solution but you can run the last deCONZ beta release without X11 with the following command

    $ DECONZ_OPTS="-platform minimal --dbg-info=1" deCONZ-autostart.sh

We don't provide a proper systemd startup script yet, so for autostart one must be implemented which calls the above command (without the --dbg-info option).

clook, 2017-07-26:
Hi @manup, works for me (no more need for a display :) )
Thanks for the great job btw! I just received my Raspbee yesterday and I use it to catch remote events (Hue dimmer) with websocket.

I will build a Docker image soon if some are interested in.

abmantis, 2017-09-21:
@clook any luck on the docker image? I'm trying to run one but I'm having some issues (GCFFlasher not working, can't find devices and zigbee channel is always zero). 
Can you post your dockerfile?

ymaclook, 2017-09-25:
Hi @abmantis, not up to date but here it is (armv7 for a Raspberry pi 2)
```
FROM arm32v7/debian:jessie-slim

ENV SW_VERSION 2.04.56
ENV DEBIAN_FRONTEND noninteractive

RUN echo 'deb http://archive.raspberrypi.org/debian/ jessie main' > /etc/apt/sources.list.d/raspi.list && \
	apt-key adv --keyserver pgp.mit.edu --recv 82B129927FA3303E && \
	apt-get update && \
	apt-get install -y curl sudo && \
	curl -O http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-${SW_VERSION}-qt5.deb && \
	(dpkg -i deconz-${SW_VERSION}-qt5.deb || true) && \
	rm deconz-${SW_VERSION}-qt5.deb && \
	apt-get install -y -f && \
	apt-get remove -y curl && \
	apt-get autoremove -y

CMD /usr/bin/deCONZ --auto-connect=1 -platform minimal --http-port=80
```

I run it with:
```
docker run -d --device /dev/ttyAMA0:/dev/ttyAMA0 --volume /root/.local/share/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ --name deconz-headless deconz-headless
```

and to use it through another container:
```
docker run -d --link deconz-headless:deconz-rest --name=deconz-freebox --restart=always deconz-freebox
```
with the following URL to connect to it (python websocket here):
```
self.ws = create_connection("ws://deconz-rest:443/websocket")
```

Hope it helps! Otherwise, check dmesg and that everything works good without container first.

jo-me, 2017-10-17:
Hi @ymaclook,
I've tried to get it to run with your suggested dockerfile and the web application starts fine. 
Initially, it shows a message that a firmware update is available.

However, the firmware update seems to fail or is not carried out at all because it only shows the "please wait" text and does not seem to do anything. After a couple of minutes the update button reappears. 

Have you tested it with the integrated firmware update mechanism?

Thanks

clook, 2017-10-19:
Hi, jo-me,

Never tried the integrated firmware update mechanism indeed. Had it well done on physical running distribution.
Did you try to run it interactive with:
```
docker run --rm -it --device /dev/ttyAMA0:/dev/ttyAMA0 --volume /root/.local/share/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ --name deconz-headless deconz-headless
```
?

FezVrasta, 2018-01-18:
Is there any progress on this? I talked with the support via email and they said I should be able to run this headlessly, but when I try to install it it asks to install Qt dependencies, that shouldn't be needed for a headless mode..

ymaclook, 2018-01-18:
Hi @FezVrasta, @manup will confirm, but I think even headless mode highly depends on some Qt libs, like a CLI Qapplication.

FezVrasta, 2018-01-18:
Thanks for the info, the fact is that it needs 250MB of additional dependencies, plus other 50 if you want the dev API...

ebaauw, 2018-01-18:
Qt has nothing to do with X-support vs headless.  The REST API plugin depends on Qt, full GUI or headless.

FezVrasta, 2018-01-18:
I hardly believe a tool of this kind needs 250MB of Qt dependencies to work, I'm sorry...

closed 2018-09-17 by manup