dresden-elektronik/deconz-rest-plugin #10: This repository updated?

jazilla, 2016-10-16:
Hi,

Based on following the instructions.

I cloned this repository and replaced the rest plugin I got with the latest deCONZ installation (2.04.18) with a fresh build from the clone. I suspect that this repo isn't updated. E.g. the 2.04.18 deCONZ installation recognize IKEA of Sweden as a manufacturer but not the fresh build from this repo.

Do you keep this repo up to date?

Also tried "git clone --branch V2_04_18 --depth 1 https://github.com/dresden-elektronik/deconz-rest-plugin.git" but that gives me build errors.

Does the instructions for this repo guarantee me to get a clone that are compatible with the latest version of deCONZ?

manup, 2016-10-19:
Hi Jazilla,

It took a bit longer to update the public repo to the latest release.
The sources are updated now to work with the instructions.

Regards

jazilla, 2016-10-19:
Hi Manup,

Thanks for the support!

I have, however, still problems following the instruction. Now I get a lot of build errors connected to lib deCONZ. Example:
de_web_plugin.cpp:7789:48: error: ‘ParamApsAck’ is not a member of ‘deCONZ’
         uint8_t apsAck = apsCtrl->getParameter(deCONZ::ParamApsAck);

Have also tested on a fresh install.

Regards

manup, 2016-10-20:
Youre right I've could reproduce the errors. The deconz-dev-*.deb file wasn't updated and missed some new definitions in the header files. In future releases the header files will be shipped within the git repo to prevent that.

The dev package is now updated and I've verified the build with a fresh Raspberry Pi image.
Can you please do the steps 3. and 4. again, the build should succeed then.

- Download deCONZ development package
    `wget http://www.dresden-elektronik.de/rpi/deconz-dev/deconz-dev-latest.deb`

- Install deCONZ development package
    `sudo dpkg -i deconz-dev-latest.deb`

jazilla, 2016-10-20:
Thanks again for the support! Now it works fine...=)

Btw, planning to add support specific to the IKEA TRÅDFRI assortment? Can see that there are some issues today such as the bri not shown correctly when reading state.

Regards

closed 2016-10-20 by jazilla

manup, 2016-10-25:
Right now we don't have the IKEA devices here for testing (not available in Germany). We're trying to get some but there is no ETA for the integration yet.