dresden-elektronik/deconz-rest-plugin #11: All devices removed

jazilla, 2016-10-20:
Hi,

From time to time when I have done updates to the rest api, replaced the library and rebooted the pi all my devices are gone. I have even to reset all settings to be able to reconnect my devices. Is this a known issue? Is there anything I can do to avoid this (close down in a proper way (how)?, save setting and re-load if problem occurs?,...)?

This is a question rather than an issue but I don't know how to label the issue *blush*.

Regards

cmwdotme, 2016-10-24:
Hello,

 I had this same issue. It turns out that for some reason the panid had changed. If you go into settings on the website and click scan for devices it will show you the "Network ID" the lights are associated with. Above the scan button you should see your Network ID. If these don't match then you will have to copy the one from the lights and hard code a pan id in the GUI. Then open the network and rescan and they should all rejoin.

manup, 2016-10-25:
As @cmwdotme  said it seems that the network parameters have been changed for some reason. Normally recompiling the plugin won't change anything of the network configuration. The configuration and parameters like ZigBee pan ID and channel are stored in the NVRAM of the RaspBee shield.

Modifications to the configuration happens in the following cases:
* WebApp > Advanced Settings > Reset Gateway
* WebApp > Advanced Settings > Load Configuration
* deCONZ > Edit > Network Settings

To further investigate the problem can you please watch the network settings parameters and tell me which ones changed without intention. For example a before and after screenshots would be helpful.

Since 2.04.18 you can also save the active configuration including network parameters as backup in the WebApp.

closed 2018-09-17 by manup