dresden-elektronik/deconz-rest-plugin #12: API documentation

jazilla, 2016-10-25:
Hi,

Is there a updated documentation available? The documentation I can find is http://dresden-elektronik.github.io/deconz-rest-doc/ and parts of the api is as far as I understand not covered.

What I am looking for is documentation of the 'rules' part. Great if I just can get an example.

Regards

manup, 2016-10-26:
Hi @jazilla,

The updated API documentation with sensors and rules is currently in review and will be released in early November together with a minor deCONZ bugfix release.

Do you have a certain use case for a rule in mind? Maybe I can provide a short example here.

Regards

jazilla, 2016-10-27:
I am looking for a way to make an "external" REST invocation or register a listener to receive sensor events in my "main" application (today using a mix of wired, wifi, zigbee etc devices) and thought that a rule action might be what I am looking for.

Btw, If you want I can send you a package with an IKEA remote and bulb.

manup, 2016-11-09:
Hey jazilla, sorry for the late response, I missed that one.
I'm afrait we don't support that kind of external REST invocation yet, to my understanding this would work like a webhook. So right now this can only be achieved by polling the sensor state and handle the update in your application.

Thanks very much but we already have ordered the IKEA lamps which should arrive soon. Partners of us has reported that there are some odds on how the IKEA lamps work. The color handling seems to be not valid ZigBee standard and device descriptors are not matching the actual device type.

jazilla, 2016-11-09:
Hi,

Thanks for the reply!

Becomes kind of strange to use polling for e.g. a motion sensor...;)

Using your software I have noticed that I do not get the correct bri and ct state from the IKEA lamps and to set both bri and ct I have to make two invocations.

Thanks again for the support!

Br/j

closed 2016-11-09 by jazilla