dresden-elektronik/deconz-rest-plugin #13: Xiaomi Smart Home Support

Hector47, 2016-11-16:
Hi,
I was wondering if the support for some of the Xiaomi sensor is coming?
Especially the motion detector and the Xiaomi Aqara Smart Light.
Do you have any ETA for it?
Thank you!

SpencerSharkey, 2016-11-30:
if it's a standard zigbee device, you can most likely get it to join your HA network on deconz.

it would be up to you to write a proper api for the device if you want to use it outside of the zigbee ecosystem of binding attributes :)

manup, 2017-03-25:
Hi, can you provide any documentation or screenshots of the clusters?
I couldn't find any useful information and they are not listet on zigbee.org.

donnib, 2017-07-24:
@manup what is the status on the Xiaomi sensors ? They just work with RaspBee or any known issues ?

runningman84, 2017-07-30:
I am also interested if the Xiaomi devices are supported I want also buy a ConBee dongle.

ebaauw, 2017-08-26:
I think we can close this one as most Xiaomi Mi and Aqara sensors and switches are now supported.  See #42, #57, #89, #138.

stefan-kaiser, 2017-09-09:
Is it possible to get the battery state, eg. for door/window sensor?
deCONZ is showing me "PowerSource: Battery" and "Power Level: 100%", but the REST-API is not aware of that!?

manup, 2017-09-09:
The state will be added in future, Xiaomi uses a non standard way to report battery status via basic cluster. From what I've seen it's a bit different for each device, so more research is needed to figure it out.

vandalon, 2018-08-28:
I have several Xiaomi motion sensors, 1 gives temp and batt. status. the rest doesn't. I noticed that some sensors just reported their battery status back after a couple of hours, but these don't except for the 1. Can it be there are different models?

marcelfischer, 2018-12-11:
Hey all,
I added the xiaomi smart home temp and humitity sensor (the round one). Its reporting values every hour and also when I press the button on the device. But not when changes appear, for example from 21C to 23C. I saw some other bugs where this behaviour was mentioned and it should work. For example: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/42#issuecomment-347793127

Any Ideas why?

Hector47, 2019-04-10:
Hello, thanks for the good work on xiaomi devices.
But do you know if the relay controller is supported in deConz? https://www.xiaomitoday.com/xiaomi-aqaratwo-way-control-module-wireless-relay-controller/ 
Thanks

elmocito, 2019-04-22:
I am interested on this one as well :-)

closed 2019-08-27 by stale[bot]