dresden-elektronik/deconz-rest-plugin #15: Plugin part of standard installation?

jsve, 2017-01-15:
I did not complete the final step in the installation manual, but i still have access to the REST plugin. Is it time for an update of the manual, or is there some reason I don't understand for cloning and installing the plugin manually?

cfrahnow, 2017-04-04:
The deCONZ software is using the file /usr/share/deCONZ/plugins/libde_rest_plugin.so to provide the REST plugin. As long as you don't remove this file (not only renaming!) the plugin will be available.
It is true, that one step is missing in the manual: You have to copy your own built "libde_rest_plugin.so" after running `qmake-qt4 && make` into /usr/share/deCONZ/plugins/ to have it acually loaded after a restart of the deCONZ software. Make sure you have a backup of the original "libde_rest_plugin.so" file in case you mess up anything!

manup, 2017-04-14:
Thanks, the new compile instructions now also reflect where to put the recompiled plugin. It's a quite rough way though, so in future there might be something like `~/.deconz/plugins` and plugins in there would automatically have a higher load priority as default plugins.

closed 2017-04-14 by manup