dresden-elektronik/deconz-rest-plugin #16: 3rd party apps cannot change color

DjangoGithub, 2017-02-13:
Hello!
3rd party apps like the original Philips Hue App can connect and also control brightness and color temperature of the lights. However changing the color does not work.
As far as I could see the 3rd party apps only set the x and y values of the color (in the http JSON Post) and not the hue and saturation. This does not seem to be properly handled by the plugin.

manup, 2017-02-14:
Hi,
we just released a new version 2.04.35 which touched some of the color handling code — the issue might be fixed at least for group commands in the update. We have to check it for single commands though.

DjangoGithub, 2017-02-14:
Hi,
thanks for the quick reply. I just checked it but the issue is not solved. 3rd Party Apps (eg. the official Philips Hue App) that do not send hue and saturation can only change the brightness and temperature.

closed 2018-09-17 by manup