dresden-elektronik/deconz-rest-plugin #19: UPnP discovery of deCONZ REST plugin

ebaauw, 2017-03-11:
In response to M-SEARCH, the deCONZ REST plugin identifies itself as `SERVER: FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0` ([upnp.cpp, line 128](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/upnp.cpp#L128)).  However in the ssdp:alive message, it uses `SERVER: FreeRTOS/6.0.5, UPnP/1.0, IpBridge/0.1` ([line 89](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/upnp.cpp#L89)).

The Philips Hue bridge includes a line `hue-bridgeid: 001788FFFExxxxxx` (the bridge's mac address on the IP network), which makes it easier to identify the bridge.  Could deCONZ use something similar?

manup, 2017-03-25:
It should be possible by using SSDP header field extensions as described in the [spec](http://www.upnp.org/specs/arch/UPnP-arch-DeviceArchitecture-v1.1.pdf).

We'll figure something out.

ebaauw, 2017-04-15:
I've reopened this issue, as I've found some more issues with deCONZ discovery, notably inconsistencies between the info provided through UPnP, through the dresden electronik portal, through the XML description and through the `/config` resource.  See the Wiki page on [Bridge Discovery](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Bridge-Discovery).

I would really like deCONZ to support the following:
- An unauthenticated GET for `/config`;
- Inclusion of `modelid` and `bridgeid` in the `/config` response (both authenticated and unauthenticated);
- Consistent values of `modelid` (`modelNumber`), `bridgeid` (`id`, `serialNumber`), `apiversion` across UPnP, the portal, the XML description, and the `/config` resource.

Also, there's a space missing behind the colon of the `GWID.phoscon.de` header field.  I'm not sure if that's required, but all the other header lines have one.

ebaauw, 2017-04-27:
I still have issue with UPnP in v2.04.40:

When doing a search for rootdevices, deCONZ responds with:
```
HTTP/1.1 200 OK
CACHE-CONTROL: max-age=100
EXT:
LOCATION: http://192.168.76.17:80/description.xml
SERVER: FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0
ST: urn:schemas-upnp-org:device:basic:1
USN: uuid:e9e25d07-1e48-404f-ae9b-aa0049c824cd::upnp:rootdevice
GWID.phoscon.de:0x00212effffxxxxxx
```
My crude UPnP message to JSON converter does a `split(': ')` per line and breaks on the missing space after `GWID.phoscon.de:`.  Moreover I expect a `ST: upnp:rootdevice` when searching for rootdevices.

Note that deCONZ does send alive messages for the rootdevice:
```
NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.76.17:80/description.xml
SERVER: FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:e9e25d07-1e48-404f-ae9b-aa0049c824cd::upnp:rootdevice
GWID.phoscon.de:0x00212effffxxxxxx
```

I'm also a bit worried that there's three different ways to get the RaspBee mac address, although they seem to yield the same result:
- `.arg(gwDeviceAddress.toStringExt()).toLocal8Bit()` for the notify;
- `.arg(gwDeviceAddress.toStringExt())` for the response; and
- `.sprintf("%016llX", (quint64)gwDeviceAddress.ext())` in rest_configuration.cpp for the value of `config.bridgeid`.

closed 2017-05-21 by ebaauw