dresden-elektronik/deconz-rest-plugin #21: ZLL Commissioning

ebaauw, 2017-03-11:
It is possible to use the RaspBee as a Router in the Hue bridge ZLL network?  I understand that it won't  actually talk to the Hue bridge.  I'd like deCONZ to discover and interact with the devices connected by to my Hue bridge, without having to connect them to a different ZigBee network.

I suppose would require adding a ZLL endpoint with a ZLL Commissioning cluster to the RaspBee, but the deCONZ Network Settings won't allow me to add an endpoint.

manup, 2017-03-25:
It is possible but there is no official guide to configure the RaspBee to join the Hue network as router. In order to let the RaspBee/ConBee join a Hue network, the settings must be adjusted:
  - *Device Type* : Router
  - *TC Link Key* : ZLL Link Key (not the ZLL Master Key). This key used to be secret but can be found in the web (hint: google peeveeone breakout breakthrough)
  - *TC Address*: 0xffffffffffffffff
  - *Network Key*: 0x00000000000000000000000000000000
  - *Predefined PANID*: disabled
  - *Static NWK Address*: checked, should be set to some arbitrary number
  - *Channel Mask*: 11, 15, 20 and 25

After saving the configuration:
  - press the **Leave** button
  - in the Hue app start automatic searching for lights (this opens the ZigBee network)
  - press the **Join** button

Connection to the hue bridge should now be established, be aware that there might be unexpected behaviour since two commissioning tools try to control the network in different ways. For example one bridge **might** delete the groups and scenes created by another one, since it doesn't know that them and assume they can be deleted.

ebaauw, 2017-03-25:
> Connection to the hue bridge should now be established

Brilliant!  I tried the (once secret) ZLL Master Key before, and failed, but the ZLL Link Key did the trick.  Many thanks!

> be aware that there might be unexpected behaviour since two commissioning tools try to control the network in different ways

For now, I'm just charting the ZigBee network using the RaspBee.

closed 2017-03-25 by ebaauw

ebaauw, 2017-03-25:
Do I understand correctly that the ZLL Commissioning cluster is used only for touch linking?  I see the Hue bridge has a client cluster and each light seems to have a server cluster.

manup, 2017-03-26:
It's mainly used for Touchlink but it's also handy to query controlled groups from light switches. ZLL devices are required to have an server cluster.

manup, 2017-04-21:
Thats correct behaviour, the RaspBee as well as any other ZLL device store all network parameters like PANID and network key which are required to operate in the network in NVRAM.

When you open the deCONZ network settings most network parameters from NVRAM are visible there.

Reconfiguring the RaspBee from router to a coordinator in the same network is not recommended, because right now the network uses a distributed trust center (ZLL) and once the RaspBee becomes a coordinator it should serve as a trust center which would confuse some devices in the distributed trust center network. So I would recommend either to don't change RaspBee to coordinator or to reset devices and rejoin them to the fresh coordinator network.

ebaauw, 2017-04-21:
> When you open the deCONZ network settings most network parameters from NVRAM are visible there.

Yes, but the Network Key shows all zeroes.

> So I would recommend either to don't change RaspBee to coordinator or to reset devices and rejoin them to the fresh coordinator network.

Thanks, understood.  I suppose the REST API wouldn't even know whether the RaspBee is a coordinator or router?  
I've got 38 lights, 10 Hue motion sensors, 6 Hue dimmer switches and 2 Hue tap switches - joining all these to a different network for testing and then joining them back to the Hue bridge before it's dark is no fun.   I'll keep the RaspBee as router for now.

oleg77s, 2017-10-30:
I can not connect the RaspBee as a Router to the Hue bridge ZLL network.
After saving the configuration:
press the Leave button 
... and  ZLL Link Key (81....B8) resets to the ZLL Master Key (5A...39)

ebaauw, 2017-10-31:
I've seen the same in recent deCONZ versions.

oleg77s, 2017-10-31:
Hmm, and what deCONZ version will work?

manup, 2017-10-31:
The problem is not the application deCONZ but recent firmware versions for RaspBee/ConBee which were modified for IKEA and OSRAM issues; sadly lost the router ability.

Following not yet official firmware might work again (but I have not testet it as router yet).

[Uploading deCONZ_Rpi_0x261a0500.bin.GCF.zip…]()

oleg77s, 2017-11-01:
My RPI system is Raspbian GNU/Linux 9.1 (stretch)
I installed GCFFlasher as discribed in the RaspBee-BHB-en.pdf:

pi@raspbee:~ $ wget http://www.dresden-elektronik.de/rpi/gcfflasher/gcfflasher-latest.deb
--2017-11-01 08:56:23--  http://www.dresden-elektronik.de/rpi/gcfflasher/gcfflasher-latest.deb
Resolving www.dresden-elektronik.de (www.dresden-elektronik.de)... 84.19.195.242
Connecting to www.dresden-elektronik.de (www.dresden-elektronik.de)|84.19.195.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 127384 (124K) [application/x-debian-package]
Saving to: ‘gcfflasher-latest.deb’
gcfflasher-latest.d 100%[===================>] 124.40K   490KB/s    in 0.3s    
2017-11-01 08:56:24 (490 KB/s) - ‘gcfflasher-latest.deb’ saved [127384/127384]

pi@raspbee:~ $ sudo dpkg -i gcfflasher-latest.deb
Selecting previously unselected package gcfflasher.
(Reading database ... 119931 files and directories currently installed.)
Preparing to unpack gcfflasher-latest.deb ...
Unpacking gcfflasher (2.10) ...
Setting up gcfflasher (2.10) ...

and got errors:

pi@raspbee:~ $ sudo GCFFlasher –f
rmmod: ERROR: Module ftdi_sio is not currently loaded
rmmod: ERROR: Module usbserial is not currently loaded
/usr/bin/GCFFlasher.bin: error while loading shared libraries: libQtCore.so.4: cannot open shared object file: No such file or directory

manup, 2017-11-01:
Sorry I should have mentioned this, on Raspbian stretch the standalone GCFFlasher package is broken, it's on the todo list.

As workaround please use GCFFlasher from the deCONZ package as follows:

```bash
$ sudo GCFFlasher_internal -f deCONZ_Rpi_0x261a0500.bin.GCF
```

**Note** deCONZ must be closed before.

oleg77s, 2017-11-01:
pi@raspbee:~ $ sudo GCFFlasher_internal -f deCONZ_Rpi_0x261a0500.bin.GCF
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
no device specified use RaspBee (/dev/ttyS0)
using firmware file: deCONZ_Rpi_0x261a0500.bin.GCF 
reset target 

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01

flashing 127162 bytes: |==============================|
verify: ....
SUCCESS
pi@raspbee:~ $ 

Thanks, @manup

oleg77s, 2017-11-01:
still  ZLL Link Key (81....B8) resets to the ZLL Master Key (5A...39)

oleg77s, 2017-11-02:
What old version of fw can i use?

mtornblad, 2017-12-03:
deCONZ_Rpi_0x26050500.bin.GCF seems to work, gets installed in /usr/share/deCONZ/firewall with the release version of deconz.