dresden-elektronik/deconz-rest-plugin #25: restrict IP on REST API plugin

NisseDILLIGAF, 2017-04-07:
Hi..
I'm running deCONZ and openHAB on the same raspberry pi.
On openHAB I have Hue Emulation, so that I can control everything on my Google Home.
I had to setup two IP's, so that both deCONZ and openHAB could show Hue devices...
The thing is that REST API plugin sets servers on all IP's... is there a way I can restrict it to one IP address?

and a side note..
I hope that deCONZ (REST API plugin) will support IKEA's trådfri devices.. including the dimmer :)
by the way, my IKEA trådfri motion sensor is recognized as a light but sets up a group that reacts to it's sensor... took a while before I figured it out :)

Thanx for a great product, and looking forward to the code refactoring... :)

manup, 2017-04-11:
I've put the item on our feature request list, it should be possible to limit the interfaces with Qt. Maybe as parameter in the configuration file.

The IKEA Trådfri sensor is now in the new beta release, you may have to delete the old one — if it's not updated automatically after resetting the device and searching for new sensors. Support is quite basic and will advance over the next releases.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases

NisseDILLIGAF, 2017-04-11:
Thanks for looking into it..
The new beta is working great with the IKEA Trådfri sensor, but I'm guessing the dimmer isn't supported yet?

closed 2017-04-12 by NisseDILLIGAF

manup, 2017-04-13:
The dimmer needs some more work and testing, it's a nasty little device — I wonder how long the battery is supposed to work.

FezVrasta, 2018-02-26:
Any update on this?