dresden-elektronik/deconz-rest-plugin #30: Lights API differences with Hue API

ebaauw, 2017-04-23:
I've finished my compatibility testing of the `/lights` resource against the Hue bridge API, see the [Wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Lights).  I've found the following differences, for which I would need to make changes to my Hue application:
1. deCONZ exposes `state.bri` for _On/off plug-in units_ (and I assume _On/off lights_) that don't support brightness.
2. deCONZ exposes `state.ct` for _Colour lights_ that don't support colour temperature;
3. deCONZ exposes `state.hue`, `state.sat`, and `state.xy` for _Colour temperature lights_, that don't support colour;
4. deCONZ exposes `state.effect` for lights that don't support colour (_On/off plug-in units_, _Dimmable lights_, _Color temperature lights_);
5. Setting `"bri": 0` turns off the light.  On the Hue API, this sets the light to minimal brightness.  Note that setting `"bri": 254` in deCONZ doesn't turn the light on.
6. deCONZ seems to convert the values of the corresponding ZigBee attributes for `state.hue`, `state.sat`, and `state.xy`, where the Hue API uses the raw values;
7. deCONZ exposes `manufacturer` instead of `manufacturername`;

manup, 2017-04-26:
Thanks a lot for the detailed tests! The falsely exposed bri, ct, hue, sat, xy and effect attributes will be fixed soon, similar to the sensor rewritten resource system in the last beta the lights resources will also be refactored.

Turing off the lights shut only be done when "on": false is given, so that will be fixed too.

> deCONZ seems to convert the values of the corresponding ZigBee attributes for state.hue, state.sat, and state.xy, where the Hue API uses the raw values;

That's more complicated, right now I don't think we will change that until it's causing errors.

> deCONZ exposes manufacturer instead of manufacturername

Tricky, we need to check if apps already using our *version* before changing this, won't be ncie to break stuff.

LunaSela, 2017-04-26:
Although it might make code less readable, it might me sensible to make behaviour switchable per API key.

So deCONZ specific apps still would get "manufacturer" and others would get "manufacturername".

Or we could add both. And deprecate the "old" one. As there are certainly more apps which expect "manufacturername", doing something might be a benefit.

manup, 2017-04-26:
Yes I think the deprecate way is the most smooth way to adjust the API

ebaauw, 2017-04-26:
Thanks @manup!

> That's more complicated, right now I don't think we will change that until it's causing errors.

Could you please elaborate (just to satisfy my curiosity).  I've always wondered why Philips chose to place the RGB to CIE 1931 conversion in apps, instead of on the bridge.  I've never understood the usefulness of `state.hue` and `state.sat` as the provided algorithms only deliver the values of `state.xy`.

> Or we could add both. And deprecate the "old" one. As there are certainly more apps which expect "manufacturername", doing something might be a benefit.
> Yes I think the deprecate way is the most smooth way to adjust the API

Agreed.  Applications should be able to handle / ignore an additional key.

ebaauw, 2017-08-08:
Status update for v2.04.62:
1. Solved.
2. Solved.
3. Solved.
4. Solved.
5. Still the case.  deCONZ actually issues an event for `state.on` (false), `state.bri` (0), and then `state.bri` (1).
6. Still the case.
7. Still the case.

@manup, what's your take on the remaining issues?

ebaauw, 2017-08-14:
> 5. Setting `"bri": 0` turns off the light. On the Hue API, this sets the light to minimal brightness. Note that setting `"bri": 254` in deCONZ doesn't turn the light on.

The light now stays on at minimum brightness, but deCONZ still reports `state.on` as `false`.  
I started sniffing the commands the Hue bridge sends, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing.  The Hue bridge actually sends _Move to level_ 0 when specifying `"bri":0`.

> 6. deCONZ seems to convert the values of the corresponding ZigBee attributes for state.hue, state.sat, and state.xy, where the Hue API uses the raw values;

Confirmed by sniffing.

> 7. deCONZ exposes manufacturer instead of manufacturername

Changed in b2f5e52.

manup, 2017-08-14:
> I started sniffing the commands the Hue bridge sends, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing. The Hue bridge actually sends Move to level 0 when specifying "bri":0.

There are two possible ZigBee commands
1) Move to Level
2) Move to Level (with on/off)

Both may carry 0 as argument but only the second will turn off the light.

The difference between `{ "on": false, "bri":0 }` and `{ "bri":0 }` should reflect which command should be send.

ebaauw, 2017-08-14:
Yes, the Hue bridge sends Move to Level when specifying bri: 0 and Move to Level (with On/Off) when specifying `{"on": false, "transitiontime": 4}` (or any value of transitiontime > 0).

ebaauw, 2018-11-23:
Let's close this for now.  I plan to do a new series of tests to compare the APIs, probably during the Xmas holidays.

closed 2018-11-23 by ebaauw