dresden-elektronik/deconz-rest-plugin #31: CLIP sensors broken?

ebaauw, 2017-04-26:
When I create a CLIP sensor, it's `config` and `state` attributes are empty.  Trying to PUT `{"on": true}` to `config` returns an error 6, parameter, on, not available, but `config.on` is created anyway.  Trying to PUT `{"flag": true}` to `state` gives no error, but nothing happens.  Trying to put a valid attribute, but for another sensor type (e.g. `state.temperature` for a CLIPGenericFlag sensor), is also accepted.

manup, 2017-04-26:
Hmm weird, *should* work, but honestly they are not testet that much by us. I put it on the list should be fixed easily.

ebaauw, 2017-04-26:
I use them extensively:
- I keep a GLIPGenericStatus sensor per room to keep track of room presence (as opposed to motion).  It's maintained through Hue bridge rules based on the motion sensors in the room _and_ in the adjacent rooms (so I can detect when some-one leaves the room).  I also use it to "disable" the motion sensor, while still being able to monitor motion.
- I also keep a CLIPGenericFlag sensor per room, as virtual master switch for the room.  Again, it's maintained through Hue bridge rules, based on the CLIPGenericStatus sensor and on the dimmer switch.  I expose the flag to HomeKit as Switch, so I can also set it directly from a HomeKit app or Siri. A second set of Hue bridge rules control the lights, based on the flag and the motion sensor's lightlevel.   I use HomeKit automation to control my Sonos speakers, based on the flag.
- Finally, I keep a CLIPTemperature and a CLIPHumidity sensor that I update from Wunderground through a little cron job (I'm Dutch - why pay for a personal weather station when I can get the same data from the Internet for free).  These are exposed to HomeKit as well.

manup, 2017-05-03:
The CLIP sensors are quite useful, I haven't realised that in first place. Maybe a automatic test script should be written, to verify that the feature works correctly for future releases?

>These are exposed to HomeKit as well.
Thats a really interesting aspect and in addition to the CLIP sensors, exposing more ZigBee devices to Homekit will make the whole system much more valuable for users. Beside IKEA devices support for example the Busch-Jaeger, GIRA and JUNG ZLL switches would be very nice

ebaauw, 2017-05-03:
> The CLIP sensors are quite useful, I haven't realised that in first place

I doubt even Philips realise that (or they would have exposed them to HomeKit themselves in the first place).

> Maybe a automatic test script should be written, to verify that the feature works correctly for future releases?

Maybe, but that's becoming a bit too much work for a hobby.  However, I've scripted my entire Hue bridge setup (over 150 rules), and plan to run that against deCONZ once the CLIP sensors are functional.  Hence my interest in mirroring the lights and devices on the Hue bridge as much as possible (until I'll have ditched the Hue bridge).  I plan to publish these scripts on GitHub, as soon as I've found a decent way to anonymise them.

> exposing more ZigBee devices to Homekit ...

I'm more than willing to help, but, again, within the limitations of a hobby.  I won't be buying devices just to support them.  Having said that, with help of some of my users, homebridge-hue does expose the FLS-PP3 to HomeKit (as a single accessory with two LightBulb services), and the dimmable light endpoint of the Busch-Jaeger switch.  I have the Trådfri remote working in my development environment, basically re-using the code for the Hue dimmer switch.  Today, I received the Ubisys D1 that I ordered for my for my dining table lamp (which has three 80W, 240V R7S halogen "bulbs" - the Busch-Jaeger is only rated for 200W, so I cannot use that).  Of course, deCONZ had better deliver, as I don't expect to be able to use the Ubisys with the Hue bridge ;-).  I'm also very interested in some ZigBee door/window sensors (looking at the NYCE sensors).

> ...will make the whole system much more valuable for users

That might be challenging:
- As I mention under **Caveats** in the homebridge-hue README, this is not really an end-user solution.  Even on the Hue developers (!) forum, homebridge is considered "hacky".  To reach a larger numbers of users, you'd have to bundle this in something like the out-of-the-box RaspBee gateway to hide the technical details.  I already received a request for a complete Raspberry Pi image.  
- The HomeKit Accessory Protocol, used by homebridge (and by any HomeKit accessory) hasn't been published by Apple.  The original paper that reversed engineered this protocol, was removed at Apple's request, see https://github.com/KhaosT/HAP-NodeJS#notes.  I wouldn't be surprised to see Apple mobilise its lawyers, or technically block homebridge, once this would attract larger numbers of users.

ebaauw, 2017-06-09:
This week at their WWDC, Apple made the HAP documentation available to all registered developers, for "personal, non-commercial use and enjoyment, and not for distribution or sale".  Looks like homebridge might be around for quite some time.

ebaauw, 2017-06-18:
> I put it on the list should be fixed easily.

Still not fixed in v2.04.45 ?

manup, 2017-06-18:
I hope they work again in 2.04.45, see commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ab9566f846f1a05d3f9d9511b4e6a3839c73644c

But it's not testet yet.

ebaauw, 2017-06-18:
Yesterday, I downloaded and installed the binary from `http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.45-qt5.deb`.  Doesn't seem to work.  I haven't reset the database, but I did create a new CLIPGenericFlag sensor.

manup, 2017-06-18:
There is a new version 2.04.46 available, to address several issues with the CLIP sensors (see commit logs). The lastupdated timestamp uses UTC now.
Still needs some more testing. :)

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.46-qt5.deb

ebaauw, 2017-06-18:
Thanks, Manuel!
Just installed 2.04.46.  Indeed `lastupdated` shows the UTC time, in the API as well as in websockets events.

When I create a CLIPGenericFlag or CLIPGenericStatus sensor, `config.on` is indeed initialised to `true`, but `state` is still empty, `{}`.  Trying to set the state now gives and error 6, `parameter, flag, not available` or `parameter, status, not available`.

ebaauw, 2017-06-18:
The problem is with the creation of the sensors.  Feeling lucky, I edited the database and added `"flag": true` and `"status":0` to the sensor states.  After that, they work as expected, including websockets notifications.

They show `config.reachable` as `false`.  The Hue bridge shows this as `true` for CLIP sensors (which is, in fact, the case as they're implemented on the bridge/gateway).

manup, 2017-06-18:
> The problem is with the creation of the sensors. Feeling lucky, I edited the database and added "flag": true and "status":0 to the sensor states. After that, they work as expected, including websockets notifications.

Indeed the creation process can do better, currently the API expects that the `state` attributes are provided in the POST request. Other than that they are only re-created while database is processed after restarting deCONZ. I'll refactor the create method to create the related `state` attributes for the specific sensors, event if not provided in the request.

>They show config.reachable as false. The Hue bridge shows this as true for CLIP sensors (which is, in fact, the case as they're implemented on the bridge/gateway).

The reachable attribute can be set via API too, I'll change the default value to true to match the hue API.

manup, 2017-06-21:
The new version 2.04.47 addresses the attribute creation and default values in commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/98988d59e5163d4808e79cbdb932f144ab1c4a42

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.47-qt5.deb

ebaauw, 2017-06-23:
Just installed v2.04.47 and tested CLIPGenericFlag, CLIPGenericStatus, CLIPTemperature, and CLIPHumidity.

The good:
- POST, GET, PUT, DELETE of CLIP sensors from the API seem to work;
- `state.lastupdated` is in UTC;
- `config.reachable` is true on sensor creation;
- Use of CLIP sensor state in a rule condition seems to work.

The bad:
- PUT of the CLIP sensor state from a rule action doesn't work - the rule doesn't get triggered (`lasttriggered` remains 0).  When leaving the conditions untouched and only changing the action to update a light state instead, the rule works as expected;
- `config.reachable` is `false` after deCONZ restart.  It remains `false` when the state is updated through the API.

The ugly:
- On sensor creation, `state` is still `{}`.  The state is populated with the appropriate state attribute (`flag`, `status`, `temperature`, `humidity`) and `lastupdated` when you PUT the state.  Apps that depend on the state to be initialised will likely fail.  Notably apps that create a companion CLIPGenericStatus sensor to a Hue dimmer switch to cycle through scenes when _On_ is pressed.
- The API also allows updating `config.reachable` of a ZigBee sensor.

manup, 2017-06-26:
Thanks again for testing, this is very helpful. There a still some things missing like CLIPLightLevel and identical default values but it's getting closer.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.48-qt5.deb

The last commits address the bad and ugly issues and are available in version 2.04.48.

`Note` This version also includes a new firmware to keep sensors working which are connected directly to the RaspBee/ConBee as child node, it can be installed from within the WebApp.

There was one issue I couldn't recreate:
> A rule that sets a light state to {"on":true,"bri":254} doesn't get triggered either. When only changing the action body to {"on":true}, the rule works as intended;

Both variants worked in my tests with a hue E27 bulb, so this needs a bit more observation.

ebaauw, 2017-06-26:
Thanks, Manuel.  I still need to do some more thorough testing on rules.  I noticed one of the commits addresses the issue of rules firing constantly when they contain only a condition on the sensor state - even before I logged this :-)

ebaauw, 2017-07-02:
> There was one issue I couldn't recreate

My bad, I made an error creating the rule, I defined `"body": {"on":"true","bri":254}` instead of `"body": {"on":true, "bri":254}`.

On v2.04.49, a rule with:
```json
    "conditions": [
      {
        "address": "/sensors/xx/state/flag",
        "operator": "eq",
        "value": "true"
      }
    ]
```
still fires continuously (~2.5 times per second?), while the sensor's `state.flag` _is_ `true`.  On the Hue bridge, it only fires when the sensor's `state.flag` _becomes_ `true` (even a PUT of `/sensors/xx/state` with body `{"flag": true}` doesn't fire the rule when `state.flag` already was `true`).

In general, rules on the Hue bridge only fire only when the last condition _becomes_ true.  Most times this would be a `dx` or `ddx` condition on `/sensors/xx/state/lastupdated`, but I have several use cases for rules that don't include such condition.

ebaauw, 2017-07-16:
> Beside that, the performance to evaluate a rule after an event has fired will be improved, it's quite slow currently.

Indeed, it takes a couple of seconds to fire a rule based on a condition using a CLIP sensor in v2.04.52.  Furthermore, the rules don't fire after restarting deCONZ.  I need to update the CLIP sensor's `state.flag` through the API to get the rule working again.

Hue bridge rules fire without noticeable delay.  My guess would be that the Hue bridge doesn't use a timer for rules, but evaluates the conditions when a state change occurs.

Using websockets proves brilliant for debugging purposes.  I even get a notification when a scene has been recalled, but only after (?!) the notification for the changes to lights and groups.

Here's the websocket log for my Washroom, which uses an Eve door sensor to set/clear the flag and gateway rules to control the light based on the flag, using a group scene recall:
```
Sun Jul 16 2017 11:01:31: {"e":"changed","id":"250","r":"sensors","state":{"flag":true,"lastupdated":"2017-07-16T09:01:31"},"t":"event"}
Sun Jul 16 2017 11:01:33: {"e":"changed","id":"251","r":"lights","state":{"on":true},"t":"event"}
Sun Jul 16 2017 11:01:33: {"e":"changed","id":"250","r":"groups","state":{"any_on":true},"t":"event"}
Sun Jul 16 2017 11:01:34: {"e":"scene-called","gid":"250","r":"scenes","scid":"1","t":"event"}
Sun Jul 16 2017 11:01:46: {"e":"changed","id":"250","r":"sensors","state":{"flag":false,"lastupdated":"2017-07-16T09:01:46"},"t":"event"}
Sun Jul 16 2017 11:01:47: {"e":"changed","id":"251","r":"lights","state":{"on":false},"t":"event"}
Sun Jul 16 2017 11:01:47: {"e":"changed","id":"250","r":"groups","state":{"any_on":false},"t":"event"}
```
Note the delay between changing the flag at 11:01:31 and 11:01:46 and changing the light and group at 11:01:33 and 11:01:47.

Here's the websocket log for my Toilet, which uses a Hue motion sensor and gateway rules to set the flag on motion detected, and clear it a minute after motion is no longer detected.  Again, a (second) set of gateway rules control the light, based on the flag:
```
Sun Jul 16 2017 11:07:42: {"e":"changed","id":"261","r":"sensors","state":{"lastupdated":"2017-07-16T09:07:42","presence":true},"t":"event"}
Sun Jul 16 2017 11:07:42: {"e":"changed","id":"262","r":"sensors","state":{"lastupdated":"2017-07-16T09:07:42","lightlevel":4040},"t":"event"}
Sun Jul 16 2017 11:07:43: {"e":"changed","id":"260","r":"sensors","state":{"flag":true,"lastupdated":"2017-07-16T09:07:43"},"t":"event"}
Sun Jul 16 2017 11:07:46: {"e":"changed","id":"261","r":"lights","state":{"on":true},"t":"event"}
Sun Jul 16 2017 11:07:46: {"e":"changed","id":"260","r":"groups","state":{"any_on":true},"t":"event"}
Sun Jul 16 2017 11:07:47: {"e":"scene-called","gid":"260","r":"scenes","scid":"1","t":"event"}
Sun Jul 16 2017 11:07:47: {"e":"changed","id":"262","r":"sensors","state":{"lastupdated":"2017-07-16T09:07:47","lightlevel":22090},"t":"event"}
Sun Jul 16 2017 11:08:02: {"e":"changed","id":"261","r":"sensors","state":{"lastupdated":"2017-07-16T09:08:01","presence":false},"t":"event"}
Sun Jul 16 2017 11:08:27: {"e":"changed","id":"261","r":"sensors","state":{"lastupdated":"2017-07-16T09:08:27","presence":false},"t":"event"}
Sun Jul 16 2017 11:08:39: {"e":"changed","id":"263","r":"sensors","state":{"lastupdated":"2017-07-16T09:08:39","temperature":2447},"t":"event"}
Sun Jul 16 2017 11:09:30: {"e":"changed","id":"260","r":"sensors","state":{"flag":false,"lastupdated":"2017-07-16T09:09:30"},"t":"event"}
Sun Jul 16 2017 11:09:32: {"e":"changed","id":"261","r":"lights","state":{"on":false},"t":"event"}
Sun Jul 16 2017 11:09:32: {"e":"changed","id":"260","r":"groups","state":{"any_on":false},"t":"event"}
```
In addition to the delay between changing the flag and the changing the light and group, note the delay between setting the presence at 11:07:42 and changing the flag at 11:07:43.  As I use a `ddx` condition with value `PT00:01:00`, there should be a one-minute delay between clearing the presence at 11:08:02 and clearing the flag at 11:09:32.  The value of presence actually wasn't changed at 11:08:27.  I think the notification was triggered by attribute reporting for the battery status.

ebaauw, 2017-07-28:
Thanks, @manup!  It looks like v2.04.57 does the trick.  Here's the Washroom log:
```
Fri Jul 28 2017 10:27:15: {"e":"changed","id":"250","r":"sensors","state":{"flag":true,"lastupdated":"2017-07-28T08:27:15"},"t":"event"}
Fri Jul 28 2017 10:27:15: {"e":"changed","id":"251","r":"lights","state":{"on":true},"t":"event"}
Fri Jul 28 2017 10:27:15: {"e":"changed","id":"250","r":"groups","state":{"any_on":true},"t":"event"}
Fri Jul 28 2017 10:27:16: {"e":"scene-called","gid":"250","r":"scenes","scid":"1","t":"event"}
Fri Jul 28 2017 10:27:23: {"e":"changed","id":"250","r":"sensors","state":{"flag":false,"lastupdated":"2017-07-28T08:27:23"},"t":"event"}
Fri Jul 28 2017 10:27:23: {"e":"changed","id":"251","r":"lights","state":{"on":false},"t":"event"}
Fri Jul 28 2017 10:27:23: {"e":"changed","id":"250","r":"groups","state":{"any_on":false},"t":"event"}
```
and the Toilet:
```
Fri Jul 28 2017 10:27:53: {"e":"changed","id":"261","r":"sensors","state":{"lastupdated":"2017-07-28T08:27:53","presence":true},"t":"event"}
Fri Jul 28 2017 10:27:53: {"e":"changed","id":"260","r":"sensors","state":{"flag":true,"lastupdated":"2017-07-28T08:27:53"},"t":"event"}
Fri Jul 28 2017 10:27:53: {"e":"changed","id":"261","r":"lights","state":{"on":true},"t":"event"}
Fri Jul 28 2017 10:27:53: {"e":"changed","id":"260","r":"groups","state":{"any_on":true},"t":"event"}
Fri Jul 28 2017 10:27:53: {"e":"changed","id":"262","r":"sensors","state":{"lastupdated":"2017-07-28T08:27:53","lightlevel":10837},"t":"event"}
Fri Jul 28 2017 10:27:54: {"e":"scene-called","gid":"260","r":"scenes","scid":"1","t":"event"}
Fri Jul 28 2017 10:27:58: {"e":"changed","id":"262","r":"sensors","state":{"lastupdated":"2017-07-28T08:27:58","lightlevel":22915},"t":"event"}
Fri Jul 28 2017 10:28:08: {"e":"changed","id":"261","r":"sensors","state":{"lastupdated":"2017-07-28T08:28:08","presence":false},"t":"event"}
Fri Jul 28 2017 10:29:09: {"e":"changed","id":"260","r":"sensors","state":{"flag":false,"lastupdated":"2017-07-28T08:29:09"},"t":"event"}
Fri Jul 28 2017 10:29:09: {"e":"changed","id":"261","r":"lights","state":{"on":false},"t":"event"}
Fri Jul 28 2017 10:29:09: {"e":"changed","id":"260","r":"groups","state":{"any_on":false},"t":"event"}
```
I'll be migrating more of my lights, sensors, switches and rules from the Hue bridge to deCONZ.  Keep you posted...

manup, 2017-07-28:
Cool glad it works, I figure there will be more adjustments to the events in near future (only fire if values actually change etc.).

ebaauw, 2017-07-28:
> I figure there will be more adjustments to the events in near future (only fire if values actually change etc.).

I don't think I've seen light or group events where the state hasn't changed, except maybe shortly after restarting deCONZ, when it's still discovering the lights.  For sensor events, at least `state.lastupdated` has changed.  For switches, it's essential that these event fires, even when `state.buttonevent` hasn't changed, otherwise you wouldn't be notified when pressing the same button for a second time.

I would really like to continue to receive events based on ZigBee attribute reporting, like the presence, lightlevel, and temperature events for the Hue motion sensor.  This would be useful for building up a history, comparable to what Eve does for their sensors.  Of course, `state.lastupdated` does change for these events, even though the lightlevel, etc don't.  Same for CLIP sensors: when I PUT the current value to `state.flag`, `state.status`, `state.temperature`, ... the attribute value doesn't change, but `state.lastupdated` does.  I would like to continue to receive these events - I run a cron job that updates a CLIPTemperature and CLIPHumidity sensor from wunderground.com every half hour (and they're great for debugging rules on the gateway!).

I would like no longer to receive the bogus notification for presence due to attribute reporting of the power cluster.  Might be challenging, as that also updates lastupdated, which I think is incorrect, as `battery` is a `config` attribute, not a `state` attribute.

ebaauw, 2017-08-02:
For a CLIPSwitch, you don't receive an event when setting `state.buttonevent` to the same value it already has even though `state.lastupdated` is updated.  See issue #75.

manup, 2017-08-02:
The following commit should fix the problem https://github.com/dresden-elektronik/deconz-rest-plugin/commit/c7c0489cf1556890765656f18ff4fc57040a6833
Will be available in version 2.04.60.

ebaauw, 2017-08-02:
Indeed, fixed in v2.04.60.  Thanks!

ebaauw, 2017-08-06:
I see a reference for CLIPPressure in the code, but it's missing from the list of supported sensor types, https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L171.  CLIPLightLevel is missing altogether.  Shall I try and create a pull request, or are you already working on these?

Related: what's the unit for `state.pressure`?  mbar?

manup, 2017-08-07:
The list was old code not needed anymore, cleaned up here: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/9f7e7437ca4539121f7e3f6bfadc8d421c82cbb8

>Related: what's the unit for state.pressure? mbar?

Good question :)

ebaauw, 2017-08-07:
OK, I can create a CLIPPressure sensor in v2.04.62 and it seems to work.  I still cannot create a CLIPLightLevel sensor, though.

manup, 2017-08-07:
>I still cannot create a CLIPLightLevel sensor, though.

Ah I totally missed that, will be added soon.

ebaauw, 2017-08-20:
I think they're fully functional now, except for `config.on`, but that also doesn't work for ZigBee sensors.  Closing this issue.

> > Related: what's the unit for state.pressure? mbar?

> Good question :)

The corresponding ZigBee attribute for the ZHAPressure's `state.pressure` is indeed in hPa (or mbar).

closed 2017-08-20 by ebaauw