dresden-elektronik/deconz-rest-plugin #34: Error handing for unsupported attributes in PUT

ebaauw, 2017-05-02:
When specifying unsupported attributes in a PUT request body, deCONZ silently ignores these attributes, whereas the Hue bridge responds with an appropriate error.  For example, when PUTing the following body to `/api/<username>/lights/1/state`:
```json
{
  "on": true,
  "unsupported": true
}
```
deCONZ responds with (my formatting)
```json
[
  {
    "success": {
      "/lights/1/state/on": true
    }
  }
]
```
whereas the Hue bridge responds with (again, my formatting):
```json
[
  {
    "error": {
      "type": 6,
      "address": "/lights/1/state/unsupported",
      "description": "parameter, unsupported, not available"
    }
  },
  {
    "success": {
      "/lights/1/state/on": true
    }
  }
]
```
When specifying only unsupported attributes, deCONZ returns no response body (not even a body with an empty array).

I found out by accident, trying to PUT `{"linkbutton": true}` to `/api/<username>/config` to unlock the gateway.  Wireshark revealed that the Wireless Light Control web app actually PUTs `{"unlock":60}` to unlock the gateway.  I probably should have RTFM...  Then again, for Hue API compatibility, the deCONZ API should accept `{"linkbutton": true}` as an alias for `{"unlock":60}` and `{"linkbutton": false}` as an alias for `{"unlock":0}`.

manup, 2017-05-05:
Good catch, I put it on the list, supporting both is fine. I wonder if `PUT linkbutton` was always there, the unlock attribute was added in the assumption that linkbutton was read-only (slapping a RTFM on my self).

ebaauw, 2017-05-05:
I think it has been.  I actually use it to expose the physical button on the Hue bridge to HomeKit as stateless programmable switch.  While my bridge is tucked away in the wiring closet, others have it in plain sight and would like to use the button for automation.  On detecting `linkbutton` has become true, I reset reset it to false, so the next button press can be detected.  I expose another (regular, stateful) HomeKit switch to set `linkbutton` to true so apps can be registered.

As deCONZ doesn't have a physical switch, I won't be exposing the stateless programmable switch, but I'm still exposing the regular switch.

ebaauw, 2018-11-23:
I cannot remove the label on this one.  Low prio, though.

closed 2019-11-25 by stale[bot]