dresden-elektronik/deconz-rest-plugin #35: Do local search for deCONZ REST Server

cfrahnow, 2017-05-03:
Currently, it is possible to search for a server simply sending a GET request to [https://dresden-light.appspot.com/discover](https://dresden-light.appspot.com/discover).

Is there a way to get the code that is actually running there? Goal is it to do the search offline.

ebaauw, 2017-05-03:
You can use UPnP to discover deCONZ locally.  It's still a bit broken, see issue #19.

manup, 2017-05-05:
The UPnP issues will be improved thanks to @ebaauw findings in the next releases. The fixes then need some testing to verify that various apps like Amazon Echo still work.

The general gateway discovery is also described in the docs:

http://dresden-elektronik.github.io/deconz-rest-doc/discovery/

closed 2018-09-17 by manup