dresden-elektronik/deconz-rest-plugin #36: RESP API doesn't find Ubisys D1 switch endpoints

ebaauw, 2017-05-04:
Today I installed the Ubisys D1 - I had to pull a neutral wire from the ceiling light point to the switch location.  The light endpoint (indeed ZHA) is recognised immediately by deCONZ.  
I added the light to the Trådfri remote group and now control my dining table lamp through that remote - very cool.

However, the two Dimmer switch endpoints of the D1 don't appear in the deCONZ REST API (v2_04_40).  I wired two pulse switches to their inputs, just to see how they'd work.  Not sure I'm gonna really need them, but I would be cool if they worked.  The first input is still bound to the light itself (factory default) and works as documented.

manup, 2017-05-05:
Yes, currently only the *lights* functionality of the Ubisys devices are supported. However all Ubisys devices have excellent support for attribute reporting and bindings, so exposing them to the realtime API should be possible. The whole switch configuration was refactored to be semi-generic, this should help to integrate the D1. No ETA but I put it on the list for a rainy weekend :-)

ebaauw, 2017-05-05:
Thanks, I'll be doing my rain dance then ;-)

ebaauw, 2017-05-06:
Somehow, the factory default binding (from the first Dimmer Switch endpoint to the Dimmable Light endpoint) got removed.  I was able to restore it manually using the deCONZ GUI, for the On/Off as well as for the Level Control cluster.  I also created binding for the second Dimmer Switch endpoint.

I cannot seem to control any other lights by creating bindings from the Dimmer Switch endpoints, though.  Would Ubisys be using custom commands (like the Trådfri remote for the Previous and Next buttons), that aren't recognised by the other lights?

ebaauw, 2017-07-14:
Any progress yet?

ebaauw, 2017-08-25:
PR #139.

I was trying to create one ZHASwitch resource with different button event ranges for the different _Dimmer Switch_ endpoints.  This would be exposed to HomeKit as two buttons with _Short Press_ and _Long Press_.  
However, I didn't get there.  Now it's two resources, one for each _Dimmer Switch_ endpoint, with button events 1002, and 1001/1003 on the first endpoint and 2002 and 2001/2003 on the second endpoint.  The distinction between _DimUp_ and _DimDown_ is lost with the current scheme (and when using one resource).

To my pleasant surprise, the deCONZ web app actually understands the two ZHASwitch resources and displays them as one device with two buttons.  @manup will the new app also?  If so, I'll probably change it to using the same button event values on both endpoints:
- 1002 for _On/Off_;
- 2001/2003 for _DimUp_;
- 3001/3003 for _DimDown_.

This would be exposed to HomeKit as 6 buttons: _On/Off 1_ and _On/Off 2_ with only _Short Press_, _DimUp 1_, _DimUp 2_, _DimDown 1_, and _DimDown 2_ with only _Long Press_.

Note to self: use the same trick as for the IKEA remote: remember 2001 (_Move Up_) vs 3001 (_Move Down_) so the right value can be issued on receiving _Stop_

ebaauw, 2017-08-26:
I'll close this one for now, but I remain curious as to how the new web app will handle the two resources.

closed 2017-08-26 by ebaauw

manup, 2017-08-27:
When suitable we combine them on application level, so that for example the Hue motion sensor is shown as one device. For that the first part of the uniqueid is used.

One of the Busch Jaeger switches has 4 ZHASwitch sensor resources and a dimmable light, here only the switches are shown combined as one device and the light is just treated as separate light.

ebaauw, 2018-10-30:
I only have a D1 and cannot test the other switches.  deCONZ now exposes a single ZHASwitch resource for the D1, with two groups in `config.groups`.
```
{
  "config": {
    "group": "230,240",
    "mode": "momentary",
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "0149ae869cb0b8f55b96aedd1f553be8",
  "manufacturername": "ubisys",
  "mode": 1,
  "modelid": "D1 (5503)",
  "name": "Dining Room Table",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2018-10-27T22:46:14"
  },
  "swversion": "20171108-DE-FB0",
  "type": "ZHASwitch",
  "uniqueid": "00:1f:ee:00:00:00:19:63-02-0006"
}
```
homebridge-hue exposes this a stateless switch with two buttons, both supporting _Single Press_ and _Long Press_.

Looking at the code, homebridge-hue assumes there's still four ZHASwitch resources for the C4.  I doubt whether this is still the case. homebridge-hue doesn't yet support the S1 nor the S2.  Please open an issue with homebridge-hue, listing the REST API resources deCONZ creates for these devices (or simply attach the debug dump file, see https://github.com/ebaauw/homebridge-hue#debug-dump-file.