dresden-elektronik/deconz-rest-plugin #37: Web Sockets not working ?

dghh, 2017-05-06:
Hi,
i´m using deconz-2.04.40 on a raspberry pi 3 with Jessie and a ZigBee addon board.

REST API is working fine (nearly like the HUE api-:) )
Webapplication is also working very fine and fast.

But, i have big problems to get anything useful out of the websockets. I´m connected to port 443 (seems to be free on my raspberry) and can connect. At least i think so...because there is no error and i get some bytes from the server.

This is my config:

**"apiversion":"1.0.0","dhcp":true,"gateway":"192.x.x.x","ipaddress":"192.x.x.x","linkbutton":false,"localtime":"2017-05-06T12:16:05","mac":"x:x:x:x:x","name":"deCONZ-GW","netmask":"255.255.255.0","networkopenduration":60,"panid":62156,"portalservices":false,"proxyaddress":"none","proxyport":0,"swupdate":{"notify":false,"text":"","updatestate":0,"url":""},"swversion":"20440","timeformat":"24h","timezone":"Europe/Berlin","utc":"2017-05-06T10:16:05","uuid":"d0cffba6-d5d0-45e0-ad64-db8330a8a90d","websocketport":443,"whitelist":{"2D6E7F6E20":{"create date":"2017-05-06T07:19:51","last use date":"2017-05-06T10:06:55","name":"deCONZ WebApp"},"B2AC314636":{"create date":"2017-05-05T12:06:46","last use date":"2017-05-06T10:15:58","name":"deCONZ WebApp"},"D2BB70681D":{"create date":"2017-05-05T12:50:00","last use date":"2017-05-06T10:16:05","name":"4711"}},"wifi":"not-installed","wifiappw":"","wifichannel":"1","wifiip":"192.168.8.1","wifiname":"Not set","wifitype":"accesspoint","zigbeechannel":15}**  

I´ve included as a test one HUE Bulb and i can switch it, dimm it etc.
But there is no output on the websocket...i´ve expected that there is some kind of JSON output when i switch the bulb on/off.

I´ve tried it with PHP/Python/HTML etc. client sockets, but nothing worked. Only some bytes after the start of the connection.

Output of the client socket (in this case it was a small python script, but looks similar if i use php) and after this nothing more...:

**--- request header ---
GET / HTTP/1.1
Upgrade: websocket
Connection: Upgrade
Host: 192.168.178.40
Origin: http://192.168.178.40
Sec-WebSocket-Key: 3i6eMLAG4U8PWC483MKIIA==
Sec-WebSocket-Version: 13
-----------------------
--- response header ---
HTTP/1.1 101 Switching Protocols
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Accept: TeOUveApLCI5/J02aLtp/ctEUNM=
Server: deconz
Access-Control-Allow-Credentials: false
Access-Control-Allow-Methods: GET
Access-Control-Allow-Headers: content-type
Access-Control-Allow-Origin: *
Date: Sa., 06 Mai 2017 10:37:13 GMT
-----------------------**

Can you please advice me (or a kind of php/python example script) how to get a correct client connection to the deconz websockets and get a JSON output if something changes on the zigbee network or in deconz server?

Regards Georg

ebaauw, 2017-05-06:
Currently, the websocket only issues events for sensors, see issue #32.

dghh, 2017-05-06:
Thank you very much ! Get gray hair in the meantime.
When can we expect to have websockets also for switches / lights ? Do you know this ?

dghh, 2017-05-06:
Just forget something. Do you have some example script how to use this websocket (beside that only sensors at the moment available)

ebaauw, 2017-05-06:
I use a very basic NodeJS script on my MacBook, just to monitor what events are being published:
```javascript
#!/usr/local/bin/node

'use strict';

const WebSocket = require('ws');
const host = '192.168.xxx.xxx';
const port = 443;

const ws = new WebSocket('ws://' + host + ':' + port);

ws.on('open', function open() {
});

ws.on('message', function message(data, flags) {
  const date = String(new Date()).substring(0, 24);
  console.log(date + ": " + data);
});
```

NodeJS doesn't seem to come with a native web sockets implementation, so you need to `npm install ws` in the directory where you place the script.  Note that `node` installs under `/usr/bin` on the Raspberry Pi, so you might need to change the first line.

To install NodeJS on macOS, just [download](https://nodejs.org/en/) and run the installer (I'm using the 6.10.3 LTS version).  For the Raspberry Pi, follow [these steps](https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions) (I'm using the 6_x version).

ebaauw, 2017-05-06:
PS. The script was inspired by the [API documentation](http://dresden-elektronik.github.io/deconz-rest-doc/websocket/).

dghh, 2017-05-06:
Hi Eric,
thanks for the script.
It seems that i have already tried it this way. I was only unsure because i've got not the expected results. Any way, again, thank you very much and let's wait for the implementation. This can be a big step in the right direction, because constantly polling hue bridges or deconz bridges is not funny.
Regards Georg

closed 2017-05-07 by dghh