dresden-elektronik/deconz-rest-plugin #38: can not compile master branch

holli73, 2017-05-12:
hello,

it tried to compile the latest commit and it looks like the deconz.h file is missing in git?

```
/home/pi/DeConz/deconz-rest-plugin
pi@rasp-eg-verteilerkasten ~/DeConz/deconz-rest-plugin $ make
make -f Makefile.Release
make[1]: Entering directory '/home/pi/DeConz/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.41\" -DGIT_COMMMIT=\"7b398f49b439114ee4a52794577c7661f771a379\" -DGW_MIN_RPI_FW_VERSION=0x260b0500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WEBSOCKETS_LIB -DQT_WIDGETS_LIB -DQT_SQL_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -isystem /usr/include -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSql -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -o release/authentification.o authentification.cpp
In file included from authentification.cpp:11:0:
de_web_plugin_private.h:24:20: fatal error: deconz.h: Datei oder Verzeichnis nicht gefunden
 #include <deconz.h>
                    ^
compilation terminated.
Makefile.Release:377: recipe for target 'release/authentification.o' failed
make[1]: *** [release/authentification.o] Error 1
make[1]: Leaving directory '/home/pi/DeConz/deconz-rest-plugin'
Makefile:34: recipe for target 'release' failed
make: *** [release] Error 2
```

i did search the complete pi/root directories but this file is not available.

best regards
holli

holli73, 2017-05-12:
hello,

from google i found

https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/6_Software/deconz-cpp-doc/index.html

but after install of this package:

dpkg -i deconz-dev-latest.deb

i got 

```
de_web_plugin.h:22:4: error: #error "The REST plugin requires at least deCONZ library version 1.3.0."
   #error "The REST plugin requires at least deCONZ library version 1.3.0."
    ^
Makefile.Release:410: recipe for target 'release/bindings.o' failed
make[1]: *** [release/bindings.o] Error 1
make[1]: Leaving directory '/home/pi/DeConz/deconz-rest-plugin'
Makefile:34: recipe for target 'release' failed
make: *** [release] Error 2
```

thanks
holli

holli73, 2017-05-12:
hello,

i figured it out

wget https://www.dresden-elektronik.de/rpi/deconz-dev/deconz-dev-2.04.40.deb

and install this one - and build worked.

but maybe interesting for others as well

sorry
holli

closed 2017-05-12 by holli73

manup, 2017-05-12:
Hi, glad it works now.
There is also this short guide on how to compile the plugin here:

https://github.com/dresden-elektronik/deconz-rest-plugin#usage

.. a bit hidden on the bottom of the page, I guess it should be linked in the REST API doc.

Rania-M, 2020-08-18:
Hi, i am re opening the subject because i too get this error, the link given above doesn't work and i have the same error : deconz.h not found ( it's not on the gir project), thank you for any given help :)

SwoopX, 2020-08-18:
Install the `deconz-dev` package via apt.
https://phoscon.de/en/conbee2/install

Rania-M, 2020-08-19:
hi @SwoopX  thank you for your answer, 
I am on a windows 10 environment, i don't find any dconz dev package for it, and i also tried to do it via docker based on an ubuntu image ( apt install deconz-dev) but still isn't working, does the deconz-dev package contains the deconz.h file ? and so where to find it ? thank you