dresden-elektronik/deconz-rest-plugin #39: 403 Fordbidden

ebaauw, 2017-05-12:
I found some more differences in error reporting between the deCONZ API and the Hue API.

When trying to create a username (POST `/api`) while the link button has not been pressed, the Hue bridge returns HTTP status code 200 OK with a body of:
```json
[{"error":{"type":101,"address":"/","description":"link button not pressed"}}]
```

When trying to create a username while the gateway is still locked, the deCONZ REST API plugin (v2.04.40) returns HTTP status code 403 Forbidden.  It still returns the error in the response body, but note the different value for `address`:
```json
[{"error":{"address":"","description":"link button not pressed","type":101}}]
```

When trying to access a resource with an invalid username (e.g. GET `/api/invalid/lights/1`), the Hue bridge returns HTTP status code 200 OK with a body of:
```json
[{"error":{"type":1,"address":"/lights","description":"unauthorized user"}}]
```

In this case, deCONZ returns status 403 Forbidden, again with an error in the response body, again with a different address:
```son
[{"error":{"address":"api/invalid/lights/1","description":"unauthorized user","type":1}}]
```

I think the values for `address` should be changed to match the Hue API.  As for the 403 status, I'm no HTTP expert so I'd better RTFM, or in this case the HTTP/1.1 spec (RFC 2616):
>  10.4.4 403 Forbidden
   The server understood the request, but is refusing to fulfill it.
   Authorization will not help and the request SHOULD NOT be repeated.
   If the request method was not HEAD and the server wishes to make
   public why the request has not been fulfilled, it SHOULD describe the
   reason for the refusal in the entity.  If the server does not wish to
   make this information available to the client, the status code 404
   (Not Found) can be used instead.

I'm not overly thrilled with the differences between the Hue and deCONZ APIs, but, looking at the spec, deCONZ behaves correctly.  
I'm not too sure about the "request SHOULD NOT be repeated" part in the second statement for the create user.  Typically, an app will want to repeat this request until the gateway has been unlocked.

murarimeena, 2020-08-06:
please provide the exact solution.

Mimiix, 2020-08-06:
@murarimeena Can you open a [new issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=User+Question&template=user-question.md&title=) for your problems?