dresden-elektronik/deconz-rest-plugin #40: deCONZ crashes on delete user

ebaauw, 2017-05-12:
When I delete a user from the whitelist (DELETE `/api/<username1>/config/whitelist/<username2>`), it no longer shows in the whitelist, but deCONZ still accepts requests for the deleted user (e.g. GET `/api/<username2>/lights/1`).  After a while, deCONZ crashes with a Segmentation fault (core dumped).  After restart, the deleted user is back in the whitelist.

Looking in the zll.db database, the users aren't delete from the `auth` table after doing the API call.  It looks like deCONZ crashes when trying to update the database, because of some other activity (in this case, discovering one of my Hue motion sensors).  Here's the output of `deCONZ --dbg-info=1`:

```
16:22:53:850 SensorNode id: 8 (Washroom Temperature) available
16:22:53:850 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0001 on endpoint 0x02
16:22:53:850 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0402 on endpoint 0x02
16:22:53:850 SensorNode id: 6 (Washroom Light Level) available
16:22:53:850 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0001 on endpoint 0x02
16:22:53:850 discard double entry in binding queue
16:22:53:850 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0400 on endpoint 0x02
16:22:53:851 SensorNode id: 7 (Washroom Motion) available
16:22:53:851 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0001 on endpoint 0x02
16:22:53:851 discard double entry in binding queue
16:22:53:851 0x0017880102xxxxxx (SML001) create binding for attribute reporting of cluster 0x0406 on endpoint 0x02
16:22:53:851 added ZCL value 0x0406/0x0000 for 0x0017880102xxxxxx
16:22:53:869 added ZCL value 0x0400/0x0000 for 0x0017880102xxxxxx
16:22:56:474 Bind response success
16:22:56:496 Bind response success
16:22:56:517 Bind response success
16:22:58:922 save zll database
Segmentation fault (core dumped)
```

I've put the gzipped core on [Dropbox](https://www.dropbox.com/s/q95aort38fl6y7n/core-deCONZ-sig11-user1000-group1000-pid761-time1494598978.gz?dl=0).

deCONZ v2.04.40 on a Raspberry Pi 3 Model B, `uname -a` reports:
> Linux pi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux

manup, 2017-05-15:
Thanks for the report and core dump, seems like I messed with iterators in C++ std containers. I'll have a look whats going on.

ebaauw, 2017-05-15:
I seems to go well when I only delete a single username.  The crash happens when I delete multiple usernames (I'm testing the creation of usernames by my homebridge plugin).

manup, 2017-06-11:
Took a while :) the last commits should fix the crash and authentication bugs

ebaauw, 2017-06-18:
Seems to be solved in v2.04.45.  Thanks.

closed 2017-06-18 by ebaauw