dresden-elektronik/deconz-rest-plugin #43: deconz on armhf

jaxer, 2017-05-19:
I am trying to set up RaspBee on ODroid-C2 (armhf) architecture. Is there any place where i can get binaries for it?

manup, 2017-05-21:
Hi, there are only binaries for the supported platforms Raspberry Pi, Ubuntu and Windows. Maybe the Raspberry .deb binaries might work?

jaxer, 2017-05-22:
It worked indeed. Problem was with dependencies and not the package itself. Thanks

closed 2017-05-22 by jaxer

esseki, 2017-11-23:
Hi @jaxer,

i'm currently trying to use Deconz with a Cubox-i (iMX6 Arm board) and i'm facing a dependency issue : 

_deconz depends on wiringpi; however:
Package wiringpi is not installed._

See #284 

This library "wiringpi" being related to the RPi GPIO, I don't expect it to be avaialbe for the Odroid C2.
So i'm just curious to know how we solved this issue to make Deconz work on Odroid C2 ?

Thank you in advance for any advice :)