dresden-elektronik/deconz-rest-plugin #47: Telekom Contact Sensor support

RobusK, 2017-05-28:
Hi,

I'm trying to get one of these to work and be visible in the REST endpoint:
https://www.amazon.de/gp/product/B01JJJ52JQ/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

What needs to be done? I can see the device in deCONZ GUI as such, but nothing in the REST. I've tried fiddling with the zcl files, but with no success

[image]

manup, 2017-05-28:
Hi, can you please provide:

* model id from basic cluster (cluster info)
* manufacturer code (node info)

I'll add them to the supported devices list, the sensor should be visible via REST API after rejoin while sensor search is active.

In the related commit you will see how its done.

RobusK, 2017-05-28:
Hi,

[image]
[image]

Perfect, thanks!

manup, 2017-05-30:
Took a while, large commit ahead ;)

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/9a0e2353f32099b8a901e2a675348fc56e53c8ea

Since the Bitron sensor is just a rebranded Climax sensor it should just work with the existing IAS Zone code.
In order to test it, please start sensor search and reset the sensor as described in the manual (often button press for 10 s). After rejoin the sensor should be discovered as presence sensor in the REST API and also trigger realtime events via web socket.

*Note:* Beside the REST API call you can start sensor search in the Web App > Settings > `Open network`

RobusK, 2017-06-05:
Hi @manup ,

I just pulled & compiled from master - sadly the sensor still doesn't show up at /api/foo/sensors :(

`{
  "1": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true
    },
    "ep": 2,
    "etag": "39b91dcb6242bbd89987c495e8725126",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "ZHALight 1",
    "state": {
      "lastupdated": "2017-06-05T22:57:06",
      "lightlevel": 7827
    },
    "swversion": "6.1.0.18912",
    "type": "ZHALight",
    "uniqueid": "00:17:88:01:02:03:af:3d-02-0400"
  },
  "2": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true
    },
    "ep": 2,
    "etag": "39b91dcb6242bbd89987c495e8725126",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "ZHAPresence 2",
    "state": {
      "lastupdated": "2017-06-05T22:57:18",
      "presence": false
    },
    "swversion": "6.1.0.18912",
    "type": "ZHAPresence",
    "uniqueid": "00:17:88:01:02:03:af:3d-02-0406"
  }
}`

manup, 2018-09-17:
Needs to be rechecked, the sensor is in the whitelist.

Closing the oldest issues for know to tidy up the tracker and duplicates in newer issues.

closed 2018-09-17 by manup