dresden-elektronik/deconz-rest-plugin #48: Schedules not working

dghh, 2017-05-29:
Hi,
i´ve created a schedule which should switch off some lights after 20 seconds, but it doesn´t switch off the lights. Also switch on the lights doesn´t work.

Although the created schedule disappears after 20 seconds in the api (as it should)
Switching the same group on/off manually works.

**Create Schedule: (with POST /api/<apikey>/schedules)**
{
    "name":"group_8",
    "description":"group_8",
    "command":{
        "address":"xxx.xxx.xxx.xx:8080/api/api_key/groups/8/action",
        "method":"PUT",
		"body": { "on": false}
    },
    "time":"PT00:00:20"}

**Answer from API:** 
[{"success":{"id":"25"}}]

**Created Schedule:**
{"25":{"autodelete":true,"command":{"address":"xxx.xxx.xxx.xx:8080/api/api_key/groups/8/action","body":{"on":false},"method":"PUT"},"description":"group_8","etag":"333d51dea86c3d6902eac1d66d21118e","name":"group_8 25","starttime":"2017-05-29T17:48:19","status":"enabled","time":"PT00:00:20"}}

By the way...websockets (motion, lights, plugs) are working very well in the beta. 
Thanks for the good job !

Regards Georg

ebaauw, 2017-05-29:
I haven't tried myself, but I think you should only specify the path in `address`, not the hostname:port.  See also: http://dresden-elektronik.github.io/deconz-rest-doc/schedules/

dghh, 2017-05-29:
Yes, a few minutes ago i´ve tried this and it works, but for instance for switching scenes or groups or lights  it also works with the hostname:port (did not try it without so far..). And i think this is the correct way, because it should be callable also from another computer than the one where dconz is running.
At least api behaviour is inkonsistent in this case.

manup, 2017-05-30:
In the current implementation the *target* host for the command is fixed as the localhost, so the gateway sends the command to itself. Therefore as @ebaauw mentioned the hostname:port should not be specified.

closed 2017-05-31 by dghh