dresden-elektronik/deconz-rest-plugin #49: Build fails for de_web_plugin.cpp

dwaldhans, 2017-05-29:
Hello,

I just tried to compile the HEAD version of the rest plugin as described in the README, but the build fails with the following error message:

```
[...]
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV6 -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.43\" -DGW_API_VERSION=\"1.0.1\" -DGIT_COMMMIT=\
"f0f3e95c4083d0408acdc2e4c0724421d8b2a35a\" -DGW_MIN_RPI_FW_VERSION=0x260b0500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"RaspBee-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WEBSOCKETS_LIB -DQT_WIDGETS_LIB -DQT_SQL_LIB -DQT
_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -isystem /usr/include -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-li
nux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSql -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf
/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -o release/de_web_plugin.o de_web_plugin.cpp
{standard input}: Assembler messages:
{standard input}:61496: Warning: end of file not at end of a line; newline inserted
{standard input}:61575: Error: unknown pseudo-op: `.uleb'
{standard input}:61170: Error: invalid operands (*UND* and .ARM.extab sections) for `-'
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Makefile.Release:500: recipe for target 'release/de_web_plugin.o' failed
make[1]: *** [release/de_web_plugin.o] Error 4
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
Makefile:34: recipe for target 'release' failed
make: *** [release] Error 2
```

Compilation of the old versions of the plugin on my old wheezy installation succeeded without any problems. I switched to jessie for the newest version just now.

Can anybody help me?

Thanks,
David

dwaldhans, 2017-06-15:
Never mind! I tried it on my other RPi and it worked there.

closed 2017-06-15 by dwaldhans

manup, 2017-06-15:
Cool that it works now, maybe it was some weird library dependency issue.