dresden-elektronik/deconz-rest-plugin #50: Websocket: 1st event seems to get lost most Time

mploetne, 2017-07-07:
Hi,
If I establish an web socket connection to the gateway all seems to be fine. however it seem that the first event after establishing a new connection get lost in most try’s.
For instance, If I switch off or on a group with two lights I get events for one of the lights and the group but the first light don’t got an event. On a second cycle all event’s are received as expected. That behaviour occurs in 9 of 10 cases. I tried it with chrome on Mac and Android, Safari and also with an client from the Chrome Webstore.
As an workaround it would be cool to emit an event to a newly connected client or better to implement the possibility to send some „Hello“ to the server and get an „World!“ back.
As I’m not an Developer, just an enthusiast, feel free to correct me. It could absolutely possible that I missed something.

mploetne, 2017-07-07:
Sorry,
forgot to state that i use deconz 2.04.50 whith the corresponding dev package and the latest git version of the plugin on rpi with the raspbee shield. i observed this behavior with all the earlier versions with ws support.

manup, 2017-07-08:
Hi I'll do some tests to figure out the problem, websocket broadcast code is quite small.  Providing request/response commands might be indeed useful beside the events.

ebaauw, 2017-07-08:
> Providing request/response commands might be indeed useful beside the events.

Please consider the security implications before going there.  I think it's already scary that one can connect to the web socket "for free", but as long as it's only reporting... ok.

manup, 2017-07-09:
Good point thats indeed a bit scary, at least there should be a initial login via api key to establish a websocket connection.

Lately I'm thinking a lot how to bake HTTPS support into REST API to improve the weak security with plain apikey. Technically it's easy — there is in fact already a HTTPS server in deCONZ since ages, but usability with self signed certificates in browsers is horrible and always looks like malware. It's a pity that private IP address ranges can't provide HTTPS for non-tech people without looking scary.

ebaauw, 2017-08-06:
> however it seem that the first event after establishing a new connection get lost

Indeed, I'm seeing the same.  E.g. when pressing the _On/Off_ button on the Trådfri remote, I only get the light and group events, not the sensor buttonevent.  Here's the log of pressing the button twice, after restarting the web socket listener:
```
Sun Aug 06 2017 16:01:12: listening on pi2.local:443
Sun Aug 06 2017 16:01:15: {"e":"changed","id":"1","r":"lights","state":{"on":true},"t":"event"}
Sun Aug 06 2017 16:01:15: {"e":"changed","id":"39119","r":"groups","state":{"any_on":true},"t":"event"}
Sun Aug 06 2017 16:01:17: {"e":"changed","id":"1","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-08-06T14:01:16"},"t":"event"}
Sun Aug 06 2017 16:01:17: {"e":"changed","id":"1","r":"lights","state":{"on":false},"t":"event"}
Sun Aug 06 2017 16:01:17: {"e":"changed","id":"39119","r":"groups","state":{"any_on":false},"t":"event"}
```

ebaauw, 2017-08-06:
It doesn't always happen - occasionally I do receive the first event.

Also, when another web socket listener is already running, the first event (for the new listener) is received by the existing listener, even when the new listener doesn't receive it.

closed 2018-09-17 by manup