dresden-elektronik/deconz-rest-plugin #51: deCONZ gateway switches since version .49 or .50 constantly the port number

dghh, 2017-07-07:
I have a serious problem with the gateway at the moment. 

Without any other changes on the raspberry the gateway switches after booting (or even while online) from port 8080 to port 80. 

It seems also that some process starts the deCONZ automatically, maybe more than onetime. 

I cannot find anything in the  usual folders that are responsible for automatical (re-)starting of processes/programs.

As i run a webserver on port 80, the webserver is not working. Also i didn´t understand why/how deCONZ can use port 80 if it is occupied by the webserver.

I have three questions:
- how can i make 100% shure that, whoever starts deCONZ, deCONZ will use port 8080  or any other but 80 ?
- what are all the folders where some software from deCONZ is installed ?
- was there a change in the last two releases that can be responsible for this behaviour ?

deCONZ was running the last 4-5 weeks without any interruption....

dghh, 2017-07-08:
I think i found the the issue here (at least i have a workaround):

I´m running deCONZ as a service in /etc/systemd/system. I´m also running ha-bridge as a service. (both on different ports) Nevertheless they are interfering with each other. 
Mostly the one or the other didn´t start or is exiting shortly after start or is having the wrong port (sometimes even 80 although this should be occupied/blocked by the webserver). 
Seems there is an issue on upnp port 1900 ? If ha-bridge comes not up there is the error that this port is blocked by another application/service.
If i disable the deCONZ service (systemctl disable ...) and start the service as last line in crontab (@reboot systemctl start ...) or completely manually, both services are working correct.
The other way round, disabling the ha-bridge service and enabling deCONZ didn´t work. Starting then ha-bridge manually after deCONZ is running a port 1900 failure occurs.

I didn´t update ha-bridge, but i updated deCONZ in the last days. Before this it was running for months without problems. (updated today also ha-bridge, but still same weird behaviour)

Maybe you can look into this, although it is running now somehow.

manup, 2017-07-08:
>I´m running deCONZ as a service in /etc/systemd/system. 

FYI deCONZ version 2.04.50 installs `/etc/systemd/system/deconz.service` it's not yet documented nor enabled and fairly basic. Intention is to provide headless support (no X needed).

deCONZ uses upnp port 1900 in shared mode so even if other services use it. As you have observed, if deCONZ is started last in row it should work fine. Looks like ha-bridge doesn't try to open the 1900 in shared mode?

In this case it should be possible to let systemd start deCONZ after ha-bridge, not sure about the details but systemd has dependency support which could be used for that.

ebaauw, 2017-07-08:
>  Intention is to provide headless support (no X needed).

Yes!

dghh, 2017-07-08:
Ok,thanks. 
Maybe the problem was, that i created this service already weeks ago and activated it. (with port 8080)
This was was possibly overwritten then.

closed 2017-07-09 by dghh