dresden-elektronik/deconz-rest-plugin #54: Setting light state to `{"on": true, "bri": 1}` turns light off

ebaauw, 2017-07-16:
Setting the light state to `{"on": true, "bri": 1}` on a Extended color light (Philips Hue) or dimmable light (innr, ubysis) turns off the light.  When setting `{"on": true}` afterwards, the light does come on at minimal brightness.

Setting only `{"bri": 1}` leaves the light on.

For Color temperature lights (OSRAM, IKEA) setting the light state to `{"on": true, "bri": 1}` leaves the light on, as expected.

wvuyk, 2018-10-11:
I also see the FLS-LP switching off when brightness is set to 1

ebaauw, 2018-11-24:
There's still something fishy going on...

Trådfri colour temperature bulb:
```
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "ct",
  "ct": 454,
  "on": false,
  "reachable": true
}
$ ph put /lights/101/state '{"on": true, "bri": 0}'
$ # Light is on
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "ct",
  "ct": 454,
  "on": true,
  "reachable": true
}
$ ph put /lights/101/state '{"on": true, "bri": 0}'
$ # Light is still on
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "ct",
  "ct": 454,
  "on": true,
  "reachable": true
}
$ ph put /lights/101/state '{"bri": 0}'
$ # Light is still on
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 0,
  "colormode": "ct",
  "ct": 454,
  "on": false,
  "reachable": true
}
$ ph put /lights/101/state '{"bri": 1}'
ph put: error: 201 parameter, /lights/101/bri, is not modifiable. Device is set to off.
$ ph put /lights/101/state '{"on": true, "bri": 1}'
$ # Light is still on
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "ct",
  "ct": 454,
  "on": true,
  "reachable": true
}
$ ph put /lights/101/state '{"bri": 1}'
$ # Light is still on
$ ph get /lights/101/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "ct",
  "ct": 454,
  "on": true,
  "reachable": true
}
```

Hue enhanced color light:
```
$ ph get /lights/226/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "xy",
  "ct": 459,
  "effect": "none",
  "hue": 13074,
  "on": false,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.510516,
    0.414973
  ]
}
$ ph put /lights/226/state '{"on": true, "bri": 0}'
$ # light is still off
$ ph get /lights/226/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "xy",
  "ct": 459,
  "effect": "none",
  "hue": 13074,
  "on": true,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.510516,
    0.414973
  ]
}
$ ph put /lights/226/state '{"bri": 254}'
$ # light is still off
$ ph get /lights/226/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "xy",
  "ct": 459,
  "effect": "none",
  "hue": 13074,
  "on": true,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.510516,
    0.414973
  ]
}
$ ph put /lights/226/state '{"on": true}'
$ # light is on
$ ph get /lights/226/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "xy",
  "ct": 459,
  "effect": "none",
  "hue": 13074,
  "on": true,
  "reachable": true,
  "sat": 205,
  "xy": [
    0.510516,
    0.414973
  ]
}
```

MatMaul, 2018-12-01:
I can confirm here with my setup, it is really annoying :)
Philips exhibits the problem but not Ikea.
Using brightness = 2 works fine.

closed 2019-04-08 by stale[bot]