dresden-elektronik/deconz-rest-plugin #55: deCONZ logs

onalzengin, 2017-07-18:
Hi,

I couldn't find where deCONZ writes log files. Can they only be seen from UI or are they written to the disk somwhere?

Thanks

manup, 2017-07-22:
deCONZ doesn't write log files to not pollute the sd-cards in long running setups.

But you can print and redirect logs to files via commandline.
For example a very noisy output can be generated as follows:

    $ deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1

Customize to your needs, for REST-API debugging the `info` and `http` options should be sufficient.

onalzengin, 2017-07-28:
Oh great! That helped a lot, thanks :)

closed 2017-07-28 by onalzengin

bramschats, 2019-08-26:
I have Deconz in Hassio running on ubuntu but I cannot seem to enter the logs using this command.

`ubuntu_docker:~$ deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1
deCONZ: command not found
ubuntu_docker:~$ `

Hope someone can tell me what to do!

bortim, 2020-06-14:
If deCONZ is used with Ubuntu and as a service you can do the following:

```
$ sudo nano /lib/systemd/system/deconz.service

ExecStart=/usr/bin/deCONZ \
        #--dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1 \
        -platform minimal \
        --http-port=80
#StandardOutput=file:/home/ubuntu/deCONZ-error.log
#StandardError=file:/home/ubuntu/deCONZ-error.log
```

If you delete the hashtag-comment, deconz is started with higher loglevel.

You can view the logifle during running the system with:
```
tail -f /home/ubuntu/deCONZ-error.log
```
@bramschats : As you see, decont is not usable everywhere and you neet the path: `/usr/bin/deCONZ`

clipse2004, 2020-11-25:
Hello, 
i tried to view log too but when i use the command above i got this error:

QCoreApplication::arguments: Please instantiate the QApplication object first
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.

what should i do to instantiate the QApplication?

Smanar, 2020-11-25:
You are using headless or desktop version ?
You need "-platform minimal" for headless

clipse2004, 2020-11-25:
well, depending on whether I want the sensors to be displayed :)

wkearney99, 2021-03-29:
if I'm running deconz on windows, where is the log?

Smanar, 2021-03-29:
On deconz / help / log