dresden-elektronik/deconz-rest-plugin #57: New Xiaomi sensors

NisseDILLIGAF, 2017-07-21:
So I bought some new Xiaomi sensors...
one _Xiaomi Aqara Temperature Humidity Pressure Sensor_ (which I got working, see below)
and one _Xiaomi Mi Smart Home Door / Window Sensors_

There is a strange thing with the door sensor... it's got Manufacturer Code 0x0000 !?
Maybe It's not an original?

anyways, this is what I got in deCONZ
http://imgur.com/h6aSOtk
http://imgur.com/IiAIAvw
http://imgur.com/EhMNszo (gif of when I move the magnet from/to sensor)

I got the Xiaomi Aqara working after adding some code...
this is the patch files for the _Xiaomi Aqara Temperature Humidity Pressure Sensor_
https://gist.github.com/NisseDILLIGAF/fca6f06153a36bcd8f8b5cf80ab81605
I hope I got it right... It looks like it's working :)
I get the three sensor values in REST

manup, 2017-07-22:
> There is a strange thing with the door sensor... it's got Manufacturer Code 0x0000 !?
Maybe It's not an original?

Some products use manufacturer code 0x0000 which sometimes means they are white label products and where not branded by buying company. 

The latest commits provide the `VENDOR_NONE` define which can be used in such cases.

>I got the Xiaomi Aqara working after adding some code...
this is the patch files for the Xiaomi Aqara Temperature Humidity Pressure Sensor
https://gist.github.com/NisseDILLIGAF/fca6f06153a36bcd8f8b5cf80ab81605
I hope I got it right... It looks like it's working :)
I get the three sensor values in REST

Cool thanks looks good, can you please open a pull request then I'll merge it into master :)

> By the way... whats new in deconz-2.04.54-qt5.deb , it's not on github...

Commits are pushed now

NisseDILLIGAF, 2017-07-22:
ok, so I tried adding `{ VENDOR_NONE, "lumi.sensor_magnet" },` to supportedDevices.
Still nothing in REST... :(
I guess I have to learn more how _ONOFF_CLUSTER_ID_ works... :)

donnib, 2017-07-24:
@NisseDILLIGAF is this the one you have ? http://www.gearbest.com/smart-light-bulb/pp_257677.html or the new one http://www.gearbest.com/access-control/pp_626703.html, do you have other Xiaomi items other than the door sensor and the temperature sensor you mentioned here in this issue ?

NisseDILLIGAF, 2017-07-25:
@donnib I have the old one, _Original Xiaomi Smart Door and Windows Sensor_ , but I don't have it working in REST yet...
The other Xiaomi sensors I have is the old temperature/Humidity sensor and the new with pressure... They are working in REST...

ebaauw, 2017-07-25:
I ordered an Aqara door/window sensor from AliExpress.  Their tracking claims it's arrived here by air, but I've yet to receive it.  Sure hope we can get it to work.  I'll post my experiences...

donnib, 2017-07-25:
@NisseDILLIGAF thanks

@ebaauw that sounds terrific, let us know how it goes.

manup, 2017-07-25:
Most (all?) of these sensors act as IAS_ZONE sensors, integration should be be possible with little adjustments.

I don't have the Xiaomi contact sensor yet but will order it soon. The temperature humidity sensor works very well till now (not sure what to do with the values though) :)

ebaauw, 2017-07-28:
Received the sensor today.  It's really cute, as small as it is.

I paired with deCONZ on the very first attempt.  I don't think deCONZ reads it correctly; it only shows a 0x0000 basic cluster and a (server!) 0x0006 On/Off cluster.  Reading the 0x0006 attributes fails.  The virtual light on the node in the deCONZ GUI blinks blue when I align or remove the magnet, so it seems to do something.

Here's a screenshot of the Node Info:
[image]

And here's a screenshot of the 0x0000 Cluster Info:
[image]

manup, 2017-07-28:
Can you please provide debug output by running deCONZ with logging enabled?

    deCONZ --dbg-aps=1 --dbg-info=1

Would be interesting to see what commands are received when magnet is moved.

ebaauw, 2017-07-28:
Wouldn't it need to client cluster to send commands?  Could it be using attribute reporting instead of commands?  I notice the 0x00 (OnOff) attribute of the 0x0006 cluster changes value immediately, when I move the magnet.  It's false when there's contact and true when I take away the magnet.  I only see only the following message in the output, each time I move the magnet:
```
16:57:38:121 APS-DATA.indication srcAddr: 0x00158d0001299e73, dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -46
```

It doesn't seem to support reading the attributes: when I try reading them, I receive an error, and afterwards all attributes are greyed.  When I double click the 0x00 attribute after restarting deCONZ, the attribute reporting window pops up, but reading the reporting config fails as well.

manup, 2017-07-28:
Ah indeed, the other Xiaomi sensors also just use attribute reporting to update their state. In this case it's a bit weird, but ok I can put something together.

Question is, what should the device be?
* ZHAPresence
* ZHAOpenClose

I figure the *ZHAOpenClose* is more appropriate in general for door and window contact sensors!?

ebaauw, 2017-07-28:
Yes, ZHAOpenClose.

I still wonder whether deCONZ reads the sensor correctly.  There is hardly any info to be found on the Internet, but this [French site](https://faire-ca-soi-meme.fr/test/2017/05/29/nouvelle-gamme-xiaomi-aqara/#Le_capteur_drsquoouverture_de_porte) lists a different device ID and a different set of clusters.  Still, no battery cluster.

manup, 2017-07-28:
For this specific device the important clusters are basic and on/off. In my experience only the Philips devices provide an usable battery cluster. Battery status for devices using CR**** batteries is usually quite inaccurate, the only useful information is if they report that battery needs to be changed. The sensors can detect voltage drops only shortly before battery dies.

ebaauw, 2017-07-30:
Sniffing the ZigBee network, indeed, the Aqara sensor sends attribute reporting for attribute 0x0000 in cluster 0x0006.

When I move the magnet like crazy, I see an attribute reporting message for attribute 0xFF01 in the 0x0000 cluster.  It's data type is 0x42 and the data contains elements 0 to 8.

abmantis, 2017-07-30:
Not sure if it helps, but have you checked the implementation for SmartThings? They connect directly to the ST hub, and the guys over there have already reverse engineered it:
https://github.com/a4refillpad/Xiaomi/tree/master/devicetypes/a4refillpad

ebaauw, 2017-08-02:
v2.04.60: deleted the sensor from the network in the deCONZ GUI.  Opened the network in the web app.  Reset the sensor.  It appears in the GUI and a resource is created in the REST API.  I even received an event for it.  The sensor is looking good, except that `state` remains empty:
```json
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "319ca4990bba547e27df402e3170e3a0",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_magnet.aq2",
  "name": "ZHAOpenClose 2",
  "state": {},
  "swversion": "3000-0001",
  "type": "ZHAOpenClose",
  "uniqueid": "00:15:8d:00:01:xx:xx:xx-01-0006"
}
```
Maybe I'm too impatient, but I noticed that the GUI only showed the Basic Cluster at first.  The OnOff cluster only appeared when I was moving the magnet.  Might it be a timing issue?

Deleted the resource, removed the sensor from the GUI and retried pairing while moving the magnet.  No change.  The GUI reacts to moving the magnet, by showing the corresponding value of the OnOff attribute in real time.

Any logging you want me to turn on?

ebaauw, 2017-08-02:
Hm, tried some more times, rebooted the Pi and I now see:
```json
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "a6948674df600ab63d112246de622493",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_magnet.aq2",
  "name": "ZHAOpenClose 2",
  "state": {
    "lastupdated": "1969-12-31T23:00:00",
    "open": false
  },
  "swversion": "3000-0001",
  "type": "ZHAOpenClose",
  "uniqueid": "00:15:8d:00:01:xx:xx:xx-01"
}
```
The REST API created a default `state` object, but no events nor changes when moving the magnet.  Also note the `uniqueid`: this time without the cluster?!

manup, 2017-08-02:
I think I found the problem, incoming on/off reports diedn't update the sensor.
New version will be uploaded in a few minutes

manup, 2017-08-02:
Updated version is up now, please download again (still version 2.04.60).

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.60-qt5.deb

ebaauw, 2017-08-02:
Timestamp 14:14 - I don't see any difference?

deCONZ-1.log is from pairing; deCONZ-2.log after restarting deCONZ.  Again, the `state` object is created on restart and the cluster is missing from `uniqueid`.

[deCONZ-1.log.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1193801/deCONZ-1.log.gz)
[deCONZ-2.log.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1193802/deCONZ-2.log.gz)

manup, 2017-08-02:
Yes timestamp 14:14, if you observe ZCL attribute reporting from the sensor in the sniffer for the onoff cluster, the `open` state should be updated and events should be thrown.

manup, 2017-08-02:
I'm not sure if the sensor creates the needed binding on its own, deCONZ doesn't create a binding yet.

ebaauw, 2017-08-02:
I forgot to run `dpkg` - see my updated comment.

ebaauw, 2017-08-02:
> I'm not sure if the sensor creates the needed binding on its own, deCONZ doesn't create a binding yet.

I think it does - the OnOff attribute in the deCONZ GUI already reflects the magnet in real-time before v2.04.60.  And, as mentioned in my corrected comment, I do get web socket notifications!

manup, 2017-08-02:
Ah okay, good that events are now received, I'll fix the `uniqueid` and `lastupdated` attributes in the next version. Thanks again for testing :)

ebaauw, 2017-08-02:
You're most welcome.

I tried some simple rules:
```json
{
  "1": {
    "actions": [
      {
        "address": "/lights/1/state",
        "body": {
          "on": true
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/2/state/open",
        "operator": "eq",
        "value": "true"
      }
    ],
    "created": "2017-08-02T13:16:42",
    "etag": "961efcc0123939d440e03e455f5d59ee",
    "lasttriggered": "2017-08-02T15:17:35",
    "name": "open",
    "owner": "philipshue",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 75
  },
  "2": {
    "actions": [
      {
        "address": "/lights/1/state",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/2/state/open",
        "operator": "eq",
        "value": "false"
      }
    ],
    "created": "2017-08-02T13:17:09",
    "etag": "67f4943040640c6d003383034b03dc11",
    "lasttriggered": "2017-08-02T15:18:33",
    "name": "close",
    "owner": "philipshue",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 400
  }
}
```
They work as expected.  Of course, the rules still get triggered continuously (see `timestriggered`), so we really need `lastupdated` (or different rule triggering logic in line with the Philips Hue bridge).

NisseDILLIGAF, 2017-08-02:
I cant get this to work with my 'original' door sensor that has Manufacturer Code 0x0000...?
I tried just adding `{ VENDOR_NONE, "lumi.sensor_magnet" },`, but it's not added to REST API...

manup, 2017-08-02:
From the screenshots it looks that the `ZigBee Node Descriptor` was not received, hence the 0x0000.
Can you please retry to add the sensor:

* if the sensor is somehow in system delete it first via REST API
* open the network under settings
* remove battery and put it in again

NisseDILLIGAF, 2017-08-02:
I cant delete it via REST because it's not there..
I've deleted it in deCONZ, removed battery and tried adding it several times... no luck.

manup, 2017-08-02:
Hm if the node descriptor is not read, the plugin will not create the API sensor.

Here are more things to check:
* Can you read the basic cluster in deCONZ?
  The sensor must be awake for that and should be triggered by moving the magnet.
* Is the firmware version 0x26110500?
  Menu: *deCONZ > Help > About*
* The sensor might be connected to an *problematic* parent, to keep things simple for setup, you can try to turn all ZigBee lights off. The sensor should then connect through the gateway directly.

NisseDILLIGAF, 2017-08-02:
- When I try to read basic cluster, the green light on the RaspBee light just stays green and nothing changes, just like the first screenshot..
- firmware is 0x26110500
- Just the lights or all ZigBee devices?

I'll try that tomorrow, it's getting late here, going up early for work..

manup, 2017-08-02:
> Just the lights or all ZigBee devices?

All devices which show as yellow node in deCONZ -- mostly this are the lights.

NisseDILLIGAF, 2017-08-03:
Ok, i turnd off all lights, but no luck... still the same when I connect..

runningman84, 2017-08-03:
I am curious to know if you share his opinion:
https://community.hom.ee/t/xiaomi-mi-smart-home-serie/1028/18

manup, 2017-08-03:
> I am curious to know if you share his opinion:
https://community.hom.ee/t/xiaomi-mi-smart-home-serie/1028/18

I totally agree that the Xiaomi sensors are very strange and anything but standard. 

Therefore I do understand their decision to not support them, same is true for Philips with IKEA, it burns a lot of time to integrate things which *should* work but don't play by the rules.

For the various reasons he mentioned deCONZ has included some quirks to learn more about devices in non standard ways. Since the sensors don't support OTA updates they will be around for some time and *should* be supported by Xiaomi gateways, hence it's okay to support them in deCONZ.

In short if devices *can* be integrated *and* there is reasonable demand, then we very likely will integrate them.

ebaauw, 2017-08-03:
> I am curious to know if you share his opinion:
https://community.hom.ee/t/xiaomi-mi-smart-home-serie/1028/18

He's right, technically, and no joy whatsoever.  As I prefer being happy over being right, I wholeheartedly applaud and support (where I can) deCONZ approach.

I'm no ZigBee specialist.  I'm not affiliated with any vendor.  I'm just a hobbyist playing with home automation.  In my experience there's no such thing as "a ZigBee standard".  Even between officially ZLL certified lights, there's noticeable differences:
- Different colour gamuts (even between different lights of the same vendor);
- Different ranges of colour temperature;
- Different numbers of groups and scenes;
- Optional (?) support for over-the-air firmware upgrade;
- Optional (?) support for ZigBee green power switches.
- Optional (?) support of ZigBee attribute reporting (push notifications from the light);
- Different power-on behaviour;
- Vendor-specific extensions (e.g. to set the power-on defaults).
While some of these difference are discoverable, some aren't.  It gets worse looking at switches and sensors.  It gets far worse looking at non-ZLL (certified) devices.

So, as a gateway/bridge, you have to make a fundamental choice:
1. Strictly adhere to the standard, supporting only the functionality that can be discovered;
2. Fully support known (whitelisted) devices, including the functionality that cannot be discovered.

In my experience, each vendor (Philips, OSRAM, innr, IKEA) takes the second approach, but only for their own devices.  They take the first approach for devices by other vendors, and seem to prefer finger pointing over providing end-user functionality.

> In short if devices _can_ be integrated _and_ there is reasonable demand, then we very likely will integrate them.

I'm very happy to see that deCONZ (or at least @manup) is taking the second approach, also for devices by other vendors (see also the last paragraph in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/20#issuecomment-289249805).  I think that's the Open Source spirit.

It would seem like Homee decided on the first approach.

NisseDILLIGAF, 2017-08-03:
- Now I deleted all files in `~/.local/share/dresden-elektronik/`
- Pulled the batteries from all devices
- changed channel to 25 _( I read that it's least interferens with wifi )_
- started deCONZ with _--dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1_ 
- and added my door sensor...

Manufacturer Code is still 0x0000 and it wont get in to rest api..

if I try to read basic cluster, RaspBee will lock up, the green light on the board will stay on and I can't do anything.
I have to shutdown and pull the power...

Here is the full log _( I hope there isn't any sensitive info in there )_
[https://pastebin.com/7ZXmDzLT](https://pastebin.com/7ZXmDzLT)
I've highligted all lines with device `0x00158D0001837326`

manup, 2017-08-03:
Cool thanks I'll check the logs to see whats happening here.

Does the sensor node still blink blue when you move the magnet (meaning deCONZ is receiving commands)?

If yes I can adjust the setup so that the sensor can be created from what is known (basic cluster modelid, mac address) without needing to read node descriptor.

NisseDILLIGAF, 2017-08-03:
No, after i get it into deCONZ the blue light stops.. but it changes attributes _true/false_ in cluster 0006 when i move the magnet

Thanx for great support!!!

NisseDILLIGAF, 2017-08-04:
Alright!!!
My Xiaomi door sensor works after commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/5c266a63809c6d2383774934b933661ec2464af9
Great work... and THANX again for great support @manup

manup, 2017-08-04:
Cool :) hopefully this will solve similar problems with other devices as well.
For now it can be closed.

I've ordered some more Xiaomi sensors to check what can be improved, the PIR is still missing.

NisseDILLIGAF, 2017-08-04:
Hmmm ... having some difficulties with my IKEA lights/motion sensors... noting is working...
I restarted my Pi and the lights are working... I had to pair one of the motion sensors again...

I think everything is working as it should now... strange?
I'll close this issue... Thanx again! This is the best support I ever had on a product!!! Great work!

closed 2017-08-04 by NisseDILLIGAF

snozzlebert, 2017-08-05:
Thanks @manup, I added my original Xiaomi Door Sensor (lumi.sensor_magnet) today without any problems.

ebaauw, 2017-08-05:
Just added support for the Aqara door sensor to homebridge-hue.  I can now use it in HomeKit!

I'd like to add support for the other Xiaomi sensors as well.  @snozzlebert, @NisseDILLIGAF, could you please post what these look like in the REST API?  What's the output of a GET of the /sensors resource?  I suppose the temperature/humidity/pressure sensor has three associated resources in /sensors (like the Hue motion sensor)?  Thanks!

snozzlebert, 2017-08-05:
Temperatue (Original Xiaomi):
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "baceb383ac8e25019e439fea82ed501e",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_ht",
    "name": "Temperature",
    "state": {
        "lastupdated": "2017-08-05T11:03:18",
        "temperature": 2411
    },
    "swversion": "3000-0001",
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:##:##:##:##-01-0402"
}
```

Humidity (Original Xiaomi, part of the temperature sensor, so it has the same prefix in uniqueid as the temperature sensor):
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "baceb383ac8e25019e439fea82ed501e",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_ht",
    "name": "Humidity",
    "state": {
        "humidity": 5624,
        "lastupdated": "2017-08-05T11:03:18"
    },
    "swversion": "3000-0001",
    "type": "ZHAHumidity",
    "uniqueid": "00:15:8d:00:##:##:##:##-01-0405"
}
```

Original Xiaomi Door/Window sensor (manufacturername and swversion are missing, maybe a bug?)
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "dd0b4d6a845420b02354e398acd57ce2",
    "modelid": "lumi.sensor_magnet",
    "name": "Door",
    "state": {
        "lastupdated": "2017-08-05T11:14:06",
        "open": true
    },
    "type": "ZHAOpenClose",
    "uniqueid": "00:15:8d:00:##:##:##:##-01-0006"
}
```

ebaauw, 2017-08-05:
Thanks, @snozzlebert.  Is this the temperature/humidity sensor, without pressure?

snozzlebert, 2017-08-05:
Correct, the 'old' sensor had no pressure sensor. After buying these sensors I found out that Aqara was the new line of these sensors. 
What is your experience with the Aqara temperature sensor? My 'original' one doesn't update very often. Maybe once in 8 hours. Does it only update on significant changes?

ebaauw, 2017-08-05:
> What is your experience with the Aqara temperature sensor?

I only have the Aqara door/window sensor.  I already get temperature through the Hue motion sensor, not sure if I'm ready to invest in more sensors, just for the added humidity (and air pressure).  I might get one for the bathroom, though.

> My 'original' one doesn't update very often. Maybe once in 8 hours. Does it only update on significant changes?

Normally, this would be configured through the ZigBee attribute reporting configuration.  In the deCONZ GUI, select the Temperature (0x0402) cluster.  Then in the _Cluster Info_ panel, double click on the Measured Value (0) attribute.  Enter the desired values and press _write config_.  The Hue motion sensor uses MinInterval: 10, MaxInterval: 300, Change: 20, meaning: report once every 5 minutes, and when the temperature changes by more than 0.2 degrees, but at most once every 10 seconds.

ebaauw, 2017-08-05:
> I'll fix the uniqueid and lastupdated attributes in the next version.

Indeed, they're fixed in v2.04.61.  Thanks, @manup!

ebaauw, 2017-08-19:
New toys from China just arrived!  Some more Xiaomi Aqara door/window sensors, a Xiaomi Aqara smart wireless switch, and two Xiaomi Aqara weather sensors.  The last two use the same CR2032 batteries as the IKEA remote and motion sensor (which you can get ridiculously cheaply at IKEA).  The door/window sensors use a smaller CR1632.

For pairing it's best first to reset the sensor (hold the reset button until the led blinks blue) and then quickly open the network.  When you briefly press the reset button, the sensors report attributes, not just for the value, but also the model and manufacturer in the basic cluster.  deCONZ creates the nodes without any issues, and the REST API plugin creates the resources after the relevant data has been read (might need to interaction with the sensor's button, to keep it awake).

The door/window sensor sends attribute reporting for open and close events.

Unlike the original Xiaomi button (see #89), the Aqara smart wireless switch only reports when releasing the button (not when pressing it), so deCONZ issues only 1002 buttonevents.  Also the 0x0006/0x0000 attribute in the deCONZ GUI doesn't change and always shows true.

As expected, the REST API plugin creates three resources for the weather sensors: ZHATemperature, ZHAHumidity, and ZHAPRessure, corresponding to the 0x0402, 0x0405, and 0x0403 clusters.  As expected, neither cluster supports _Read attributes_, but all three send _Report attributes_ commands.   For 0x0402, it's attribute _Measured value_ (0x0000), an int16, with the measured temperature in 0.01 degrees Celcius.  For 0x0402, it's attribute _Measured value_ (0x0000), an uint16, with the measured humidity in 0.01 %.  For 0x0405, it's three attributes: _Measured value_ (0x0000), an int16 with the measured pressure in hPa (or mbar); _Scale_ (0x0014), an int8, with value 255 (0xff - I think that's -1?); and _Scaled value_ (0x0010), and int16 which is 10x the _Measured value_, or rather _Measured value_ seems to be trunc(_Scaled value_  * 10^(_Scale_)).  Not sure how accurate these sensors are, so I wouldn't bother with anything but _Measured value_.

I don't get web socket notification for the ZHAPressure, and the state object in the sensor resource remains empty.  There's no `Added ZCLvalue 0x0403/0x000` message in the log.  After restarting deCONZ, `state.pressure` is populated, but remains 0.  `state.lastupdated` is 1969-12-31T23:00:00.

deCONZ reads the ZigBee attribute as uint16 instead of int16, https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L3325.  @manup, could that be the cause?   
Related: I see `state.temperature`, `state.humidity`, and `state.pressure` are all defined defined as int32 in deCONZ (https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/resource.cpp#L96), as are `state.buttonevent` and `state.status`.  I don't suppose that would be an issue though, as temperature and humidity work.

ebaauw, 2017-08-20:
> What is your experience with the Aqara temperature sensor? My 'original' one doesn't update very often. Maybe once in 8 hours. Does it only update on significant changes?

Looking at the web socket notifications, the Xiaomi Aqara weather sensor (temperature, humidity, pressure) sends updates throughout the night.
```
Sun Aug 20 2017 02:23:26: /sensors/346/state: {"lastupdated":"2017-08-20T00:23:26","temperature":1504}
Sun Aug 20 2017 02:23:26: /sensors/347/state: {"humidity":8326,"lastupdated":"2017-08-20T00:23:26"}
Sun Aug 20 2017 02:56:06: /sensors/346/state: {"lastupdated":"2017-08-20T00:56:06","temperature":1454}
Sun Aug 20 2017 02:56:06: /sensors/347/state: {"humidity":8428,"lastupdated":"2017-08-20T00:56:06"}
Sun Aug 20 2017 03:42:21: /sensors/346/state: {"lastupdated":"2017-08-20T01:42:21","temperature":1432}
Sun Aug 20 2017 03:42:21: /sensors/347/state: {"humidity":8466,"lastupdated":"2017-08-20T01:42:21"}
Sun Aug 20 2017 04:27:15: /sensors/346/state: {"lastupdated":"2017-08-20T02:27:15","temperature":1411}
Sun Aug 20 2017 04:27:15: /sensors/347/state: {"humidity":8448,"lastupdated":"2017-08-20T02:27:15"}
Sun Aug 20 2017 05:24:54: /sensors/346/state: {"lastupdated":"2017-08-20T03:24:54","temperature":1383}
Sun Aug 20 2017 05:24:54: /sensors/347/state: {"humidity":8471,"lastupdated":"2017-08-20T03:24:54"}
Sun Aug 20 2017 06:48:51: /sensors/346/state: {"lastupdated":"2017-08-20T04:48:51","temperature":1385}
Sun Aug 20 2017 06:48:51: /sensors/347/state: {"humidity":8597,"lastupdated":"2017-08-20T04:48:51"}
Sun Aug 20 2017 07:22:13: /sensors/346/state: {"lastupdated":"2017-08-20T05:22:13","temperature":1424}
Sun Aug 20 2017 07:22:13: /sensors/347/state: {"humidity":8540,"lastupdated":"2017-08-20T05:22:13"}
Sun Aug 20 2017 07:39:18: /sensors/346/state: {"lastupdated":"2017-08-20T05:39:18","temperature":1475}
Sun Aug 20 2017 07:39:18: /sensors/347/state: {"humidity":8550,"lastupdated":"2017-08-20T05:39:18"}
Sun Aug 20 2017 07:55:23: /sensors/346/state: {"lastupdated":"2017-08-20T05:55:23","temperature":1511}
Sun Aug 20 2017 07:55:23: /sensors/347/state: {"humidity":8465,"lastupdated":"2017-08-20T05:55:23"}
Sun Aug 20 2017 08:14:20: /sensors/346/state: {"lastupdated":"2017-08-20T06:14:20","temperature":1562}
Sun Aug 20 2017 08:14:20: /sensors/347/state: {"humidity":8344,"lastupdated":"2017-08-20T06:14:20"}
Sun Aug 20 2017 08:37:57: /sensors/346/state: {"lastupdated":"2017-08-20T06:37:57","temperature":1613}
Sun Aug 20 2017 08:37:57: /sensors/347/state: {"humidity":8114,"lastupdated":"2017-08-20T06:37:57"}
Sun Aug 20 2017 08:53:41: /sensors/346/state: {"lastupdated":"2017-08-20T06:53:41","temperature":1663}
Sun Aug 20 2017 08:53:41: /sensors/347/state: {"humidity":7948,"lastupdated":"2017-08-20T06:53:41"}
Sun Aug 20 2017 08:58:53: /sensors/346/state: {"lastupdated":"2017-08-20T06:58:53","temperature":1716}
Sun Aug 20 2017 08:58:53: /sensors/347/state: {"humidity":7890,"lastupdated":"2017-08-20T06:58:53"}
Sun Aug 20 2017 09:07:06: /sensors/346/state: {"lastupdated":"2017-08-20T07:07:06","temperature":1766}
Sun Aug 20 2017 09:08:57: /sensors/346/state: {"lastupdated":"2017-08-20T07:08:57","temperature":1819}
Sun Aug 20 2017 09:08:57: /sensors/347/state: {"humidity":7544,"lastupdated":"2017-08-20T07:08:57"}
Sun Aug 20 2017 09:09:38: /sensors/346/state: {"lastupdated":"2017-08-20T07:09:38","temperature":1872}
Sun Aug 20 2017 09:09:38: /sensors/347/state: {"humidity":7473,"lastupdated":"2017-08-20T07:09:38"}
```
Note that `pressure` is missing, since deCONZ doesn't update this attribute.  It looks like the sensor is smart enough to send multiple reports once it's awake (unlike the Hue motion sensor, which uses separate scheduling for `presence`, `lightlevel` and `temperature`.  I see no pattern in the timing, indeed, it would seem it only sends reports once a threshold for change has been crossed.  Note that at 9:07, only the temperature is sent - I assume bacause there was no change in humidity.

> Normally, this would be configured through the ZigBee attribute reporting configuration

This seem to be hard-coded in the sensor (and, in fact in all Xiaomi devices I've seen) - they don't support reading not writing the attribute reporting configuration.

As to the precision of these sensors: I placed one in the bathroom, where I have a Hue motion sensor measuring temperature, and an old-fashioned, unconnected, consumer-grade, digital thermo/hygrometer.   I placed the other one on my balcony, comparing it to the temperature and humidity I pull from Wunderground.  The Aqara's temperature seems to be a bit high (1-2 ºC), the humidity seems more or less accurate (within 2%).  
Then again, I never had two Hue motion sensors report exactly the same temperature, even when placing them right next to each other.  "With one watch a man knows what time it is, with two he's never sure".

ebaauw, 2017-08-20:
> I don't get web socket notification for the ZHAPressure, and the state object in the sensor resource remains empty. There's no Added ZCLvalue 0x0403/0x000 message in the log. After restarting deCONZ, state.pressure is populated, but remains 0. state.lastupdated is 1969-12-31T23:00:00.

That's because the attributes for 0x0403 are missing from general.xml, see #127.

Hypfer, 2017-10-14:
I can't get the Xiaomi Mi Smart Home Door / Window Sensors to work with DeCONZ 2.04.81

[image]

manup, 2017-10-14:
Please try again with 2.04.82, will be available later today.

manup, 2017-10-14:
2.04.82 is up:
http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.82-qt5.deb

Hypfer, 2017-10-14:
I found out why it didn't work: You need to to connect both parts of the sensor while adding it to the network otherwise it will fail.