dresden-elektronik/deconz-rest-plugin #58: No support for time-based conditions in rules?

ebaauw, 2017-07-22:
deCONZ returns a condition error (607) when trying to create a rule with the following condition:
```json
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T07:00:00/T23:30:00"
    }
```
The Philips Hue bridge allows this kind of time-based condition, see [the API spec](https://developers.meethue.com/documentation/rules-api).  This condition causes the rule to fire daily at 07:00:00 and evaluates to true from 07:00:00 to 23:00:00.  Similarly a condition
```json
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T23:30:00/T07:00:00"
    }
```
fires at 23:00:00 and evaluates to true until 07:00:00 the next day.

manup, 2017-07-22:
The implementation for `/config/localtime` will be available in the next days, we need it quite urgent too in order to implement proper motion sensor controls :)

ebaauw, 2017-08-05:
In v2.04.61, I can create a rule with a condition on `localtime`, but it fires constantly while the condition is true, instead of only when the condition becomes true.

ebaauw, 2017-08-08:
> it fires constantly while the condition is true, instead of only when the condition becomes true.

Should be covered by #98, so I'll close this issue.

closed 2017-08-08 by ebaauw