dresden-elektronik/deconz-rest-plugin #59: No rule condition on light or group state?

ebaauw, 2017-07-22:
deCONZ returns a Condition error (607) when trying to create a rule with a condition on a light state:
```json
    {
      "address": "/lights/101/state/on",
      "operator": "eq",
      "value": "false"
    }
```
or on a group state:
```json
    {
      "address": "/groups/0/state/any_on",
      "operator": "eq",
      "value": "false"
    }
```

While we're on the subject of group state: the Hue bridge also provides an attribute `state.all_on` for groups, which, unsurprisingly, is `true` when all lights in the group are on.

manup, 2017-07-23:
Not yet, groups and lights aren't yet part of the resource system. Implementation will arrive in the next releases.

> While we're on the subject of group state: the Hue bridge also provides an attribute state.all_on for groups, which, unsurprisingly, is true when all lights in the group are on.

Same here, but I have to admit that I'm a bit skeptical about the usefulness, some lights lie about their state :) My IKEA tuneable white E27 often says it's `off` after power-up while it's clearly `on`.

So in rules I would never rely on these values, could be right, could be wrong.

ebaauw, 2017-08-05:
In v2.04.61, I can create a condition on a group's `state.any_on`, but the rule doesn't get fired.

closed 2018-06-12 by ebaauw