dresden-elektronik/deconz-rest-plugin #61: Any hope to make the deconz run as a service without a desktop ?

donnib, 2017-07-27:
Hi,
I guess this is not entirely related to this plugin but i have to ask since i read many places that people does not like to use this because of this fact which is a shame because i believe it's a great product. I just bought one but i also think this is a major problem.

Here are links from people about this :

https://www.amazon.com/gp/customer-reviews/R3NB12924ND4YO/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B00E6300DO

https://community.home-assistant.io/t/deconz-raspbee-integration/22432/5

https://community.home-assistant.io/t/zigbee-bulb-control/224/4

Any take on this ? @manup 

/donnib

manup, 2017-07-27:
Yes the topic is quite a thing, see also

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/9

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/51

> FYI deCONZ version 2.04.50 installs `/etc/systemd/system/deconz.service` it's not yet documented nor enabled and fairly basic. Intention is to provide headless support (no X needed).

The systemd script allows to run deCONZ without X desktop as systemd service. It's only a semi solution since the usual X libs must be installed.

donnib, 2017-07-27:
@manup sorry for not searching well enough about this. I'll give it a go. If i am using the Raspbee Gateway image can i just do : sudo update-rc.d deconz defaults and have it start up automatically without a display ?

manup, 2017-07-27:
No problem, the infos on this topic are not that easy to find I'm thinking of creating a wiki page for that. 

The default RaspBee Gateway image starts deCONZ and X. Both must be disabled first and than the service must be enabled via `$ systemctl enable deconz`.

I'll write a more detailed howto later on ...

donnib, 2017-07-27:
Ok, so i have to figure out how to do following as i am sitting with a brand new image installed as we speak and wish to get this running up now, i got my RaspBee today so i am exited to get my IKEA remotes to work ;)

> Both must be disabled first 

Can you guide me how to do above ?

Jey-Cee, 2017-07-31:
Hi,  
i am looking for Linux x86/64bit version of deconz v2.04.50. But can not found a Download link.
Can any one give me a link?

svenove, 2017-10-16:
Any new info on how to enable this?

manup, 2017-10-16:
There is a basic system script included in the recent beta versions.
To enable deCONZ at boot time without X:

    $ systemctl enable deconz

svenove, 2017-10-16:
@manup Thanks, I tried it but it failed (With error 217/User). Found out why - I'm running a "vanilla" Ubuntu 16.04 and I don't have a "pi"-user. I replaced the "User=pi" with "User=root", I then was able to start the service, but all devices/groups where missing! So I replaced "root" with my normal username (which I have been running deconz under manually), but then it wouldn't start again. So I added "/usr/bin/sudo" on the "ExecStart"-line and was able to start the service - BUT all my lights are just named "Light x" and not the names I had set for them. Stopping the service and running " sudo /usr/bin/deCONZ -platform minimal --http-port=80" manually made the devices/groups to reappear.

Any idea on what I should do to be able to run as a service, keeping the configuration I've already done?

mariusmotea, 2017-10-16:
If a command is working properly when is executed directly, but fail when is executed by a service (cron, systemd, etc) most likely the issue is related to environment. Try to setup same environment before you start deconz. Maybe deconz saved some files somewhere is your home directory.

manup, 2017-10-16:
> I'm running a "vanilla" Ubuntu 16.04 

Ah I see, yes currently the script doesn't work for Ubuntu this will follow in future releases.

To get devices and groups back, the files located in  `<non-root-user>/.local/share/dresden-elektronik/deCONZ/` must be copied into the respective directory in /root.

IcarusR, 2017-10-22:
@manup I am trying to get deconz to run on a headless machine. As I don't have user pi I have changed the user name in 
`/etc/systemd/system/deconz.service`  
I can not login to the web page as new user, what password do I use ?? If I login as delight with password delight none of my lights are there.
Files are present at 
`<non-root-user>/.local/share/data/dresden-elektronik/deCONZ/`
If I start deconz manually with 
`sudo /usr/bin/deCONZ -platform minimal --http-port=8090` 
When I login with default credentials my lights are there.

Any ideas ??

Cheers

svenove, 2017-10-22:
@IcarusR: I believe I had similar issues. I ended up using the root-user... I believe the problem is that deCONZ must be run with more than just “regular” privileges - that’s why you run it with “sudo” when running it with your regular user. 
It might work to prefix with “/bin/sudo” in ExecStart, but I’m not sure. Worth a shot?

manup, 2017-10-29:
deCONZ itself doesn't need superuser rights it only requires access to the RaspBee/ConBee device to work. We are also working to get rid of all `sudo` uses in the current *deCONZ-autostart.sh* script, as this is not a clean solution.

**Note** The systemd script assigns Linux capabilities to let it run on ports < 1024 as normal user.

ooii, 2018-03-13:
Hi, I'm completely new to deConz after using the Hue bridge for a couple of years.
I'm running the deConz app on a **headless** Ubuntu with no issues but I can only access the webapp and Phoscon.
But, for now, I cannot find any doc about using these two apps and about configuring the sensors and dimmers. 
Is the GUI required for that? Can we configure dimmer's buttons using these two web-based apps?
Thanks.

Kane610, 2018-03-13:
@ooii add lights to a group you can put rules and schedules and sensors to groups. You can also add lights to remotes groups and they will be controllable by the remote.

@manup @donnib This topic should be closed.

closed 2019-03-30 by stale[bot]

ssbarnea, 2019-12-05:
I discovered that very ugly side of deconz just after I received my ConBee 2 stick and to be honest I am not considering returning the stick. As of 2019, I do not find acceptable user experience to be forced to configure X11 and use to configure an app which is not even packaged as a proper linux service. I had very different hopes: TUI and WEB interfaces and installable as a service using system packages.

At least the part about being available using a package-repository was done and that is a very good start. Still `systemctl | grep deconz` seems to return nothing even today. 

```
# deCONZ<tab><tab>
deCONZ                        deCONZ-configure-wifi.sh      deCONZ-init.sh                deCONZ-setup.sh               deCONZ-update.sh
deCONZ-autostart.sh           deCONZ-homebridge-install.sh  deCONZ-otau-dl.sh             deCONZ-startstop-wifi.sh      deCONZ-WIFI2.sh
deCONZ-check-wifi.sh          deCONZ-homebridge.sh          deCONZ-run.sh                 deCONZ-update2.sh             deCONZ-WIFI.sh
```
This looksmore like a desktop app, not a service.

ooii, 2019-12-05:
You do not need to have X11, just start it with `-platform minimal` option. And you can start it as a service with no issue. You just need to have the right file at the right place. Mine is at `/lib/systemd/system/deconz.service` and here is the content:
```
[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service

[Service]
User=1000
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target
```
And I (re)start it with `service deconz start` and `service deconz restart`.
I hope that helps.

J4nsen, 2020-05-06:
I'm with @ssbarnea  on this. It is annoying that deCONZ pulls GUI libs into a X-free system. Just look at the dependencies the binary is linked against:

        libdeCONZ.so.1 
        libsqlite3.so.0 
        libQt5Widgets.so.5 
        libQt5Gui.so.5 
        libQt5SerialPort.so.5
        libQt5Network.so.5
        libQt5Core.so.5
        libpthread.so.0
        libstdc++.so.6
        libm.so.6
        libgcc_s.so.1
        libc.so.6
        libdl.so.2
        libGL.so.1
        libpng16.so.16
        libharfbuzz.so.0
        libz.so.1
        libudev.so.1
        libicui18n.so.60
        libicuuc.so.60
        libdouble-conversion.so.1
        libglib-2.0.so.0
        libGLX.so.0
        libGLdispatch.so.0 
        libfreetype.so.6
        libgraphite2.so.3 
        librt.so.1
        libicudata.so.60
        libpcre.so.3
        libX11.so.6
        libxcb.so.1
        libXau.so.6
        libXdmcp.so.6 
        libbsd.so.0

StephanStS, 2024-01-08:
Maybe it helps some of the readers: I also struggled with the deconz.service file, especially when using a Conbee III.
The file can be found there: `/lib/systemd/system/deconz.service`

The deCONZ command line options are describe there: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters

I changed the deCONZ command line in my deconz.service in my Debian environment from 
`ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80`
to
`ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 --device=/dev/ttyUSB0 --baudrate=115200`