dresden-elektronik/deconz-rest-plugin #62: Sensors in Device WebApp

wizkidorg, 2017-07-28:
I was crudely able to get the Temperator & Humidity sensors working on the webapp.
Question, will you be moving to Bootstrap 3? (the icons do not have a temperature icon, and the glyphlings icons are in the wrong folder, they need to be in /bootstrap/img/ not /img/ 

This is what I did to make the Temp/Humidity sensors display on the device page: (changes to devices.js)

Around Line 463:
`
            // sensor already exists. Add endpoint.
            for (d in devices) {
                if (devices[d].uniqueid.indexOf(uid[0]) === 0) {
                    devices[d].ep = devices[d].ep + "," + device.id + "-" + uid[1];
                    if (device.state.humidity) devices[d].state['humidity'] = device.state.humidity;
                    if (device.state.temperature) devices[d].state['temperature'] = device.state.humidity;
                    break;
                }
            }
` 

This was needed since the xiaomi sensors show up as 2 devices but your code only detects the first sensor. This adds the second sensor to the device, the name only shows the first sensor but at least it adds it to the array.

and Around Line 510:
`
        } else if (device.type === "ZHATemperature" || device.type === "ZHAHumidity") {
                // Temperature/Humidity Sensor
                deviceIcon = "<span class=\"icon-home icon-white\">";
                rowtxt += '<tr class="light_item" style=""><td style="width:12%">' + deviceIcon + '</td><td style="width:10px;"></td>';
                rowtxt += '<td>' + device.name;
                if (device.state.temperature)   rowtxt += ' Temp: '+ (parseInt(device.state.temperature/100) * 9 / 5 +32).toFixed(2).toString() + "&deg;F";
                if (device.state.humidity)      rowtxt += ' Humidity: '+ (parseInt(device.state.humidity/100)).toFixed(2).toString()+"%";
                rowtxt += '</td>';
                rowtxt += "<tr><td colspan=\"4\">&nbsp;</td></tr>";
        }
`

This adds it to the device page and does Celsius to Fahrenheit conversion, and humidity to percent.

I would use git and "attempt" to submit my changes, but i can't figure GIT out without totally breaking things. I attached the devices.js file also.

[devices.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1181630/devices.zip)

manup, 2017-07-28:
> Question, will you be moving to Bootstrap 3? (the icons do not have a temperature icon, and the glyphlings icons are in the wrong folder, they need to be in /bootstrap/img/ not /img/

We did :) a new version of the app will be available as private beta preview in a few days. It uses Glyphicons from Bootstrap very sparingly, but instead has new dedicated icons as SVG.

Here is a screen of a combined sensor view for *Philips hue Motion Sensor*, it looks very similar for Xiaomi sensors. The sensor view does refresh in realtime due the events from the websocket.

[image]

wizkidorg, 2017-07-28:
Perfect! <Where do I signup?? :) > Will the hue hub device, when connected
to something like Smartthings show the sensors also?

closed 2018-09-17 by manup