dresden-elektronik/deconz-rest-plugin #64: Hue tap differences

ebaauw, 2017-07-28:
Hue tap seems fully functional in deCONZ.  I get websocket notifications:
```
Fri Jul 28 2017 11:06:14: {"e":"changed","id":"306","r":"sensors","state":{"buttonevent":34,"lastupdated":"2017-07-28T09:06:14"},"t":"event"}
Fri Jul 28 2017 11:06:20: {"e":"changed","id":"306","r":"sensors","state":{"buttonevent":16,"lastupdated":"2017-07-28T09:06:20"},"t":"event"}
Fri Jul 28 2017 11:06:21: {"e":"changed","id":"306","r":"sensors","state":{"buttonevent":17,"lastupdated":"2017-07-28T09:06:21"},"t":"event"}
Fri Jul 28 2017 11:06:22: {"e":"changed","id":"306","r":"sensors","state":{"buttonevent":18,"lastupdated":"2017-07-28T09:06:22"},"t":"event"}
```

The Philips Hue API doesn't show `config.reachable` for a Hue tap:
```json
{
  "config": {
    "on": true
  },
  "manufacturername": "Philips",
  "modelid": "ZGPSWITCH",
  "name": "Downstairs Tap",
  "state": {
    "buttonevent": 18,
    "lastupdated": "2017-07-23T20:55:25"
  },
  "type": "ZGPSwitch",
  "uniqueid": "00:00:00:00:00:xx:xx:xx-f2"
}

```

The deCONZ API does, and it's always `false`.  Also, the `swversion` seems to be made up:
```json
{
  "config": {
    "on": true,
    "reachable": false
  },
  "ep": 242,
  "etag": "07ddc2724df0a31a8c7e0c2e19b13774",
  "manufacturername": "Philips",
  "modelid": "ZGPSWITCH",
  "name": "Upstairs Tap",
  "state": {
    "buttonevent": 34,
    "lastupdated": "2017-07-28T08:59:25"
  },
  "swversion": "1.0",
  "type": "ZGPSwitch",
  "uniqueid": "00:00:00:00:00:xx:xx:xx-f2"
}
```
I don't use `swversion`, but I do check `config.reachable` and set `Status Fault` in HomeKit when it's `false`.  I'll probably change my code to ignore this attribute for ZGPSwitch sensors (and for CLIP sensors, as it's always `true` for these).

manup, 2017-07-28:
Guess it's best to adjust it to the hue implementation, since *reachable* attribute doesn't make sense for ZGP devices.

ebaauw, 2017-07-28:
Will do.  Thanks.

closed 2017-07-28 by ebaauw