dresden-elektronik/deconz-rest-plugin #65: Install deCONZ and development package for x64

holli73, 2017-07-28:
hello,

where do i find the devpacke for x64 linux?
i can compile on my pi3 but for my x64 with the conbee i can not compile without this libs - where can this libs be found?

thanks
holli

manup, 2017-07-31:
Hey holli,

the x64 are not updated that frequently, in fact only stable versions until now. We will release beta versions with dev-packages for Linux x64 soon — like all `2.04.x0` version releases.

manup, 2017-08-07:
Beta version 2.04.62 is noch available for Ubuntu.

https://www.dresden-elektronik.de/deconz/ubuntu/beta/

holli73, 2017-08-07:
@manup,

cool - i did load it and was able to compile on x64 linux.

now my complete house is running on bleeding edge of deconz **gg** 

thanks
holli

ebaauw, 2017-08-07:
> Beta version 2.04.62 is noch available for Ubuntu.

Seems to be working like a charm on the Parallels virtual machine I setup for BitCatcher (see #69).

I had to `sudo apt-get install qt5-default libqt5sql5 libqt5websockets5-dev libqt5serialport5-dev` before it would run.

holli73, 2017-08-19:
hello,

are there any new files for x64 (2.04.66) available?

thanks
holli

manup, 2017-08-19:
Too fast :) they will be uploaded tomorrow with an updated x64 package for Ubuntu.

manup, 2017-08-20:
Version 2.04.67 for Ubuntu is online now:
https://www.dresden-elektronik.de/deconz/ubuntu/beta/

holli73, 2017-09-01:
hello,

latest commit breaks my build on x64?

```
g++ -c -m64 -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.72\" -DGW_API_VERSION=\"1.0.2\" -DGIT_COMMMIT=\"d63849c559278346df62106d82bead3713416bfa\" -DGW_MIN_RPI_FW_VERSION=0x26160500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o release/de_web_plugin.o de_web_plugin.cpp
de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::delayedFastEnddeviceProbe()’:
de_web_plugin.cpp:8548:23: error: ‘WARMUP_TIME_S’ was not declared in this scope
     if (getUptime() < WARMUP_TIME_S)
                       ^
de_web_plugin.cpp: In member function ‘void DeRestPlugin::checkZclAttributeTimerFired()’:
de_web_plugin.cpp:9773:30: error: ‘WARMUP_TIME_S’ was not declared in this scope
         if (d->getUptime() < WARMUP_TIME_S)
                              ^
Makefile.Release:547: recipe for target 'release/de_web_plugin.o' failed
```

best regards
holli

ebaauw, 2017-09-01:
It breaks on the Raspberry Pi as well.  You need to change `WARMUP_TIME_S` to `WARMUP_TIME` or wait for @manup to push the changes for 2.04.73.

manup, 2017-09-01:
Sorry, I forgot to push the fixes, commits are pushed now

closed 2017-09-01 by manup

summer69guy, 2018-08-21:
Does this work on Debian?