dresden-elektronik/deconz-rest-plugin #66: Gateway gets stuck and doesn't respond

onalzengin, 2017-07-28:
Hi,

When I turn on/off a light by clicking on the bulb icon on deCONZ web app or on Home Assistant (home-assistant.io), sometimes deCONZ gets stuck and doesn't respond to further commands for that bulb for some time. When it's stuck, it prints out this log repeatedly with request number changed:
`delay sending request 64 to 0x000b57fffe2a41fa`. The bulbs I'm trying to control are IKEA Trådfri bulbs.

Thanks

manup, 2017-07-30:
The message says that there are open requests (waiting for ACK) to the light and hence further commands are delayed. I could recreate this situation in a quick test, needs some more debugging and sniffing.

I've also seen that the IKEA lights sometime report on/off state wrongly and will ignore some commands send by the WebApp, like `Move to level with OnOff`, until plain on/off commands are send.

onalzengin, 2017-07-30:
I'm using a plugin called Fluxer with Home Assistant which changes temperature of the lights following sun. It updates lights every 30 seconds with 30 seconds transition time. Could it be that bulbs are blocked during that transition time?

I also have Philips Hub and bulbs that are connected to Home Assistant and they work fine. I'll try to add Philips bulbs to deCONZ gateway to see whether this is an issue with bulbs or gateway.

closed 2017-10-21 by onalzengin