dresden-elektronik/deconz-rest-plugin #67: schedules use UTC instead of localtime

ebaauw, 2017-07-29:
On the Hue bridge, the `time` attribute for schedules has been deprecated in favour of `localtime`:
> time | string | 1.0 | Time when the scheduled event will occur. Time is measured in the bridge in UTC time (1.1). Either time or localtime has to be provided.DEPRECATED  This attribute will be removed in the future. Please use localtime instead.

> localtime | string | 1.2.1 | Local time when the scheduled event will occur.Either time or localtime has to be provided. A schedule configured with localtime will operate on localtime and is returned along with the time attribute (UTC) for backwards compatibility.

deCONZ still uses `time` and doesn't accept `localtime` when creating a schedule through the API. I'd hate to have to change my schedules twice a year to cope with the changes to/from daylight saving time...

Also I noticed my (recurring) schedules show `autodelete` as true in deCONZ.  On the Hue API, this attribute doesn't show on recurring schedules:
> autodelete | bool | 1.3 | If set to true, the schedule will be removed automatically if expired, if set to false it will be disabled. Default is true. Only visible for non-recurring schedules.

ebaauw, 2017-07-29:
Some more schudule blues:
- Changing the state of a CLIP sensor from a schedule doesn't work.  It's accepted on creating or changing the schedule, but the command isn't executed;
- Changing the time of a recurring schedule, e.g. from `"W127/T10:00:00"` to `"W127/T11:00:00"` between 10:00:00 and 11:00:00 (UTC) doesn't seem to cause the schedule to fire again at 11:00:00.  I probably need to do some more testing...

ebaauw, 2017-07-30:
OK, with `deCONZ --dbg-aps=1 --dbg-info=1` I see the following (filtered for the relevant messages):
```
11:19:59:795 schedule 4 diff 43801, 
11:19:59:795 schedule 5 diff 1, 
11:20:00:791 schedule 4 diff 43800, 
11:20:00:791 schedule 5: Test trigger
11:20:00:792 schedule 5 body: {"flag":true}
11:20:00:792 schedule was neigher light nor group request.
11:20:01:792 schedule 4 diff 43799, 
11:20:02:791 schedule 4 diff 43798, 
11:20:03:791 schedule 4 diff 43797, 
11:20:04:791 schedule 4 diff 43796, 
```
The message, `schedule was neigher light nor group request` speaks for itself, I'm afraid.

Apparently deCONZ is counting down to the schedule execution time, but only for schedules that haven't yet fired today.  Afterwards, I changed the time of /schedules/5, and deCONZ starts counting down again.  However, no trigger message when it's time:
```
11:29:59:790 schedule 4 diff 43201, 
11:29:59:790 schedule 5 diff 1, 
11:30:00:791 schedule 4 diff 43200, 
11:30:01:791 schedule 4 diff 43199, 
11:30:02:790 schedule 4 diff 43198, 
11:30:03:791 schedule 4 diff 43197, 
```

For completeness, here's the schedule:
```json
{
  "autodelete": false,
  "command": {
    "address": "/api/xxxxxxxxxxxx/sensors/102/state",
    "body": {
      "flag": true
    },
    "method": "PUT"
  },
  "description": "Test",
  "etag": "17a86ba226d79bf298e9d32a451fd69a",
  "name": "Test",
  "status": "enabled",
  "time": "W127/T09:20:00"
}
```

manup, 2017-07-30:
Thanks for looking into this, schedules will be reviewed in a few days to reflect latest resource system improvements.
I'll forward the issues, `localtime` support will be implemented makes a lot of sense.

dghh, 2017-09-01:
Any news to this? Seems the schedule still uses not the local,time.

snozzlebert, 2017-10-06:
I saw this commit ba1ca1a2ee8c9c2f725e6f57d9ce108567e172f1, so probably in the next version?
Right in time before October 29th ;)

snozzlebert, 2017-12-20:
This is't fixed (at least not for recurrent schedules).

ebaauw, 2017-12-20:
Did you re-create the schedule?

snozzlebert, 2017-12-21:
I'm not able to create a new schedule with ```localtime``` (with POST on /schedules)

Possible due to this check https://github.com/dresden-elektronik/deconz-rest-plugin/blob/de9853c693d22e65e4ab95a7544e9af7d84d7e5b/rest_schedules.cpp#L714

I get:
```
[
    {
        "error": {
            "address": "/schedules",
            "description": "missing parameters in body",
            "type": 5
        }
    }
]
```

ebaauw, 2017-12-21:
If specify both `time` and `localtime` it works:
```
  "4": {
    "activation": "start",
    "autodelete": false,
    "command": {
      "address": "/api/username/sensors/103/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    },
    "description": "Night On",
    "etag": "381f6d3c3fe99b35732f1cb171ee5a87",
    "localtime": "W127/T23:30:00",
    "name": "Night On",
    "status": "enabled",
    "time": "W127/T23:30:00"
  }
}
```

closed 2018-06-12 by ebaauw

Thomas-Vos, 2019-01-15:
Why was this issue closed? Schedules can still only be created when a `time` attribute is provided https://github.com/dresden-elektronik/deconz-rest-plugin/issues/67#issuecomment-353278540.

The following POST request to `/schedules`:
```
{
	"name": "Test",
	"localtime": "2019-01-15T17:51:00",
	"command": { ... }
}
```
Returns this error:
```
[
	{
		"error": {
			"address": "/schedules",
			"description": "missing parameters in body",
			"type": 5
		}
	}
]
```
If I add a `time` attribute to the body, no error is returned and the schedule is created successfully. Why is the `time` attribute required if `localtime` is provided?