dresden-elektronik/deconz-rest-plugin #68: Sensor state not stored to sqlite database

snozzlebert, 2017-07-29:
I own the 'original' Xiaomi temperature and humidity sensor (not the Aqara). State changes of the temperature and humidity are not stored in the sqlite database. So when I restart deCONZ the REST API returns the temperature and humidity of the moment at which I added the sensor to deCONZ.

E.g., state before restarting deconz:

```
"state": {
    "lastupdated": "2017-07-29T10:01:04",
    "temperature": 2350
}
```

state after restarting:

```
"state": {
    "lastupdated": "2017-07-23T18:23:53",
    "temperature": 2531
}
```

I think the same happens with my Hue Tap.

manup, 2017-07-30:
The `sensor/state` is not stored in database on each change in order to not stress the sd-card with frequent writes. We may change this so that state updates are stored in larger intervals.

ebaauw, 2017-07-30:
I think I noticed the same for light states.

deCONZ should be able write the states to the database on exit, without stressing the SD card.  The periodic database writes would be needed only if deCONZ crashes.

snozzlebert, 2017-08-05:
It looks like this is solved in 93001ebdd5334b7b8e24a529f704ac33800a0b19
I can confirm state changes are eventually saved to the database.

closed 2017-08-05 by snozzlebert