dresden-elektronik/deconz-rest-plugin #69: How to install BitCatcher on Linux

ebaauw, 2017-07-29:
Any help on installing BitCatcher on Linux is greatly appreciated.

I got a ConBee USB stick and a new RaspBerry Pi 3.  I successfully flashed the ConBee with `BitCatcher_Firmware_V2_07_03.GCF` using the GCFFlasher_internal that comes with deCONZ.  I downloaded BitCatcher from https://digital.luxoft.com/bitcatcher/ and try to install it by running
```
$ java -version
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)
$ java -jar Downloads/Sniffer_1.0.1_public/pc/BitCatcher_Setup_1.0.0.jar
```
The installation fails with an error popup:
```
Error executing
/bin/chmod a+x /home/pi/Luxoft/BitCatcher/sniffer
```
The error is to be expected, as the mentioned file doesn't exist.  The installer did create the path and a bunch of other files.  Running the installer with `sudo` installs to `/usr/local/Luxoft` instead, but reports the same error.

manup, 2017-07-30:
Not sure if BitCatcher runs on ARM, I'm using it on Arch Linux (x64) and Win7.

After installation in Linux it can be started with `sudo ./sniffer`, I have installed it as normal user in `~/opt/BitCatcher`.

ebaauw, 2017-07-30:
Ok, thanks, I'll try an Arch Linux VM...

Do I need to install a driver for the ConBee or should it be recognised spontaneously?

manup, 2017-07-30:
Ubuntu should work too, extra drivers are not needed but I think the /dev/ttyUSB0 should be put in the group of the user to get permission to it.

ebaauw, 2017-07-30:
Ubuntu does work ;-)  I copied the network key from deCONZ and see a whole load of packages.  I don't suppose there's a tutorial somewhere?

I tried a filter on the messages sent by the Trådfri remote, based on `nwkHeader->srcShortAddr`.  This still results in half a dozen to a dozen messages per button event.  I see the Remote sends a unicast to the light (router) it's connected to (green line in deCONZ GUI).  Then the light broadcasts the message, and that broadcast is repeated by other routers.  I tried also filtering on `macHeader->srcShortAddr`, but then a lot of messages aren't captured.

closed 2017-07-30 by ebaauw

ebaauw, 2017-07-30:
For those who want to try, here's the steps I took:
- Create a new Ubuntu VM under Parallels for mac - just selected the predefined Ubuntu config;
- Connect the ConBee (flashed with the BitCatcher firmware) to the VM.
- Install Parallels Tools (prompted automatically);
- Install Java: `sudo apt-get install default-jre`;
- Install BitCatcher: `sudo java -jar BitCatcher_Setup_1.0.0.jar`;
- Add user to `/dev/ttyUSB0` group: `sudo adduser parallels dialout`;
- Run BitCatcher: `/usr/local/Luxoft/BitCatcher/sniffer`;
- In BitCatcher, add a device `DE Sniffer` with port `/dev/ttyUSB0` and default communication settings;
- Connect the device (little icon next to the select device dropdown);
- Select the ZigBee channel deCONZ is using;
- Under _Preferences_ (gear icon), select _Core Security_ and add the network key deCONZ is using.