dresden-elektronik/deconz-rest-plugin #71: How do i pair a Philips Hue Iris lamp with RaspBee ?

donnib, 2017-07-30:
I tried off / on couple of times while network open but nothing, i also tries scan and then turn the lamp on close to the gateway but nothing.

ebaauw, 2017-07-31:
Try powering the other lights and ZigBee devices off, and try to use touchlink to reset the Iris.

Is this the friends of Hue Iris, or an older model?

donnib, 2017-07-31:
What is touchlink ? Is that the remote that came with it ? I don't think i have the remote :( or at least we can't find it.

This is an old model my dad had laying around, i remember pairing it with the Hue bridge for long time ago.

ebaauw, 2017-07-31:
In the deCONZ web app under _Settings_.

donnib, 2017-07-31:
Ohh i see, yes i tried that (maybe i didn't have all lights off).

manup, 2017-07-31:
For Touchlink to work it's important that the lights are very close to the gateway. Also a current firmware must be installed. If the Web App shows update firmware the firmware is too old (like shipping firmware).

>This is an old model my dad had laying around, i remember pairing it with the Hue bridge for long time ago.

Than it's probably not a ZLL certified one (labeled Friend of Hue) and won't work. Prior to hue Philips had some products which were using ZigBee but not ZigBee Light Link as per standard, these only work with the hue bridge. Same is true for the round Philips color remote (sadly because it's a cool device).

donnib, 2017-07-31:
@manup and @ebaauw so i did actually got TouchLink to work, the device showed up, i taped the blink it blinked, i clicked the Reset and i could see it did something to the lamp then i turned it off and opened network but it doesn't join, i guess this is were the fun stops ?

manup, 2017-07-31:
Yes it seems in this case its one of the older devices which won't play with non hue gateways.

donnib, 2017-07-31:
That's a shame :( it's a pretty cool light though.

closed 2017-08-01 by donnib