dresden-elektronik/deconz-rest-plugin #72: What's the best way to name devices ?

donnib, 2017-08-01:
Hi,
I tried to rename the node but nothing happened so i guess the item doesn't like that. Is there another way i can set a name on a device so when i use the REST api i have that name ?

manup, 2017-08-01:
The easiest method is to use the Web App to rename lights. Renaming sensors will be supported soon but is available via REST API now.

donnib, 2017-08-01:
I did rename thru Web App but it's not reflected in the REST API, maybe after reboot ? I did however succeed with renaming thru REST API and that worked so i guess as a start that's good.

ebaauw, 2017-08-01:
I don't think the web app has any storage of its own, so any change you make there is using the REST API.

The deCONZ GUI uses it's own names.  Sometimes it allows setting the name of a node through the _Node Info_ panel works, sometimes it doesn't.  Sometimes it takes over the name from the REST API, sometimes it doesn't.  It stores the names in a json file, `.local/share/dresden-elektronik/deCONZ/session.default`, along with the coordinates of the nodes on the screen.  Sometimes shutting down deCONZ and editing the file works, sometimes it doesn't...  I still see some of my Hue motion sensors named after the ZHATemperature resource which I cannot seem to change.

manup, 2017-08-01:
> I did rename thru Web App but it's not reflected in the REST API

Please press enter after setting the light name, otherwise it won't be applied (This is fixed in the new version).

closed 2017-08-04 by donnib

KennethLavrsen, 2021-04-17:
@manup it is 2021 - what it the new version you are talking about where the renaming in the Deconz app is fixed? I have never seen it work ever.

manup, 2021-04-20:
The fix was related to the web app (the old one) not deCONZ. I think it's time to remove the input field in deCONZ since it wouldn't allow renaming separate resources. Originially the field had the purpose to change the User Descriptor of devices (a 16 char array) but this isn't supported by almost all devices nowadays.