dresden-elektronik/deconz-rest-plugin #73: Daylight sensor?

ebaauw, 2017-08-01:
Does deCONZ support the Daylight sensor?  I see references in the code for `config.lat`, `config.long`, `config.sunriseoffset`, and `config.sunsetoffset`, but I don't see the sensor under the `/sensors` resource in the REST API.  Do I need to create it myself?

manup, 2017-08-03:
The daylight sensor is not supported yet but we will implement it in near future, because it's required in a feature we want to add to the app :)

ebaauw, 2018-04-07:
Daylight support has been added, thanks!

How about adding `state.dark`, which would be true from DL_DUSK to DL_DAWN?

I do have some small issues with the `state.status`, but nothing that couldn't be solved in documentation:
-  The attribute name is misleading as it holds the last event, rather than the status.  Took me some time to figure this out.  Using the event makes a lot of sense for triggering rules; using the status makes more sense for reporting/viewing.  I have no problem with a REST API client translating the last event into status (except for `state.daylight` and `state.dark`).
- There are inconsistencies in the namings of the #defines, DL_GOLDENHOUR1 is actually the _end_ of the morning golden hour, whereas DL_GOLDENHOUR2 is the _start_ of the evening golden hour.  Both sound more like status than events;
- Similarly, actual sunrise is at DL_SUNRISE_START, whereas actual sunset is at DL_SUNSET_END.  `state.daylight` reflects this correctly.
- I assume NIGHT_START and NIGHT_END are linked to the astronomical dusk and dawn?
```
Apr  6 06:29:52 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T04:29:52","status":130} // DL_DAWN
Apr  6 07:04:52 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T05:04:52","status":140} // DL_SUNRISE_START
Apr  6 07:08:32 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T05:08:32","status":150} // DL_SUNRISE_END
Apr  6 07:50:02 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T05:50:02","status":160} // DL_GOLDENHOUR1
Apr  6 13:44:02 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T11:44:02","status":170} // DL_SOLAR_NOON
Apr  6 19:37:52 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T17:37:52","status":180} // DL_GOLDENHOUR2
Apr  6 20:19:32 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":true,"lastupdated":"2018-04-06T18:19:32","status":190} // DL_SUNSET_START
Apr  6 20:23:02 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T18:23:02","status":200} // DL_SUNSET_END
Apr  6 20:58:12 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T18:58:12","status":210} // DL_DUSK
Apr  6 21:40:52 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T19:40:52","status":220} // DL_NAUTICAL_DUSK
Apr  6 22:27:32 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T20:27:32","status":230} // DL_NIGHT_START
Apr  7 01:43:42 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-06T23:43:42","status":100} // DL_NADIR
Apr  7 04:57:22 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-07T02:57:22","status":110} // DL_NIGHT_END
Apr  7 05:44:32 pi2 dc_eventlog[720]: /sensors/1/state: {"daylight":false,"lastupdated":"2018-04-07T03:44:32","status":120} // DL_NAUTICAL_DAWN
```

References: 
- https://en.wikipedia.org/wiki/Twilight
- https://en.wikipedia.org/wiki/Golden_hour_(photography)

manup, 2018-04-12:
> How about adding state.dark, which would be true from DL_DUSK to DL_DAWN?

Does the `state.dark` also need offset attributes or is it automatically the inverted `state.daylight` with its offsets?

> I do have some small issues with the state.status, but nothing that couldn't be solved in documentation:

Agree, internally we started to work on the documentation hopefully it will be more clear how the daylight sensor works.

> * The attribute name is misleading as it holds the last event, rather than the status. Took me some time to figure this out. Using the event makes a lot of sense for triggering rules; using the [...]

Yes and no, `state.status` is kind of event-ish but also reflects the current status/phase (state might have been a better name) which is valid until the next value is set.

In rules the `state.status` in combination with `state.lastupdated` and operator `ddx` might be useful.

> * There are inconsistencies in the namings of the #defines, DL_GOLDENHOUR1 is actually the end of the morning golden hour, whereas DL_GOLDENHOUR2 is the start of the evening golden hour. Both sound [...]

When you look at the *getDaylightTimes()* function I think this can be improved since it's all angle based, so the correct golden hour start/end points can be added to the array and other values might be adjusted/switches to better reflect the real world events.

```C++
    std::vector<TimePin> times;
    // sun times configuration (angle, morning name, evening name)
    times.push_back({-0.833, "sunriseStart",  DL_SUNRISE_START, "sunsetEnd",    DL_SUNSET_END});
    times.push_back({-0.3,   "sunriseEnd",    DL_SUNRISE_END,   "sunsetStart",  DL_SUNSET_START});
    times.push_back({-6,     "dawn",          DL_DAWN,          "dusk",         DL_DUSK});
    times.push_back({-12,    "nauticalDawn",  DL_NAUTICAL_DAWN, "nauticalDusk", DL_NAUTICAL_DUSK});
    times.push_back({-18,    "nightEnd",      DL_NIGHT_END,     "nightStart",   DL_NIGHT_START});
    times.push_back({6,      "goldenHour1",   DL_GOLDENHOUR1,   "goldenHour2",  DL_GOLDENHOUR2});
```

>* I assume NIGHT_START and NIGHT_END are linked to the astronomical dusk and dawn?

Yes it's pointed to the 18° angle like described here https://en.wikipedia.org/wiki/Twilight

ebaauw, 2018-04-22:
> Does the `state.dark` also need offset attributes or is it automatically the inverted `state.daylight` with its offsets?

After playing with the Daylight sensor for a while, I find that when using `status`, `sunriseoffset` and `sunsetoffset` seem superfluous.  I set both 0, so that `daylight` reflects time from sunrise (start) to sunset (end).  I've created some rules using `status` values less than 125 and greater than 205 to emulated `dark` from dawn to dusk.

I think it might make sense to have `tholddark` and `tholdoffset` equivalents, so specify with `status` values are considered `daylight` and which are `dark`, instead of using a time-based offset.  Not sure if it's worth the trouble, though - I'm happy using sunrise-sunset and dusk-dawn.

For my Hue motion sensor rules, I typically turn lights on when dark is true and turn them off when daylight is true - during "twilight" (both daylight and dark are false), I leave the lights as they are.

closed 2018-06-12 by ebaauw