dresden-elektronik/deconz-rest-plugin #74: Use groups or no groups, performance ?

donnib, 2017-08-02:
So i am slowly progressing in making an Home Assistant component that supports hue motion sensor, ikea lights and ikea remotes. I am debating whether i need to import groups from deConz since i can group lights in HASS. Might i run in a performance problem when the user clicks and remote button -> websocket event sent -> HASS reacts -> tells deconz to turn on 10 different lights ? I mean will the user see the lights changing from off to on one at a time or will it happen instantaneously ? I am aware that HASS performance plays a part. What's your take on this ? Which way is best to go ? I could try without first and see how it works.

ebaauw, 2017-08-02:
I spend a lot of effort optimising homebridge-hue to work well with HomeKit groups.  I also support Hue bridge (or deCONZ gateway) groups in homebridge-hue.  Here's my experience:

- When using HomeKit groups, HomeKit sends an event for each characteristic (power, brightness, hue, saturation) changed for each light (actually, I think HomeKit sends one HAP message with all the changes, but homebridge breaks this up into these events, so homebridge-hue receives up to four events per light).  I collect the events per light, and send one API request per light to the Hue bridge.  With many lights (I tested up to six), you see the lights changing independently.  According to the Philips Hue documentation, this will result in up to three ZigBee unicast messages per light (On/Off, Level Control, Hue/Saturation).  
So for six lights: 1 HAP message, 24 HAP events, 6 bridge requests, 18 unicast ZigBee messages.

- When using Hue bridge groups, I expose the group as a single light to HomeKit.  Again, I collect the events into one API request.  The lights react as one.  According to the Philips Hue documentation, this will result in up to three broadcast messages.  
So for six lights: 1 HAP message, 4 HAP events, 1 bridge request, 3 broadcast ZigBee messages.

As a final remark, the [Philips Hue documentation](https://developers.meethue.com/things-you-need-know) states:
> Number of commands you can send per second
You can send commands to the lights too fast. If you stay roughly around 10 commands per second to the /lights resource as maximum you should be fine. For /groups commands you should keep to a maximum of 1 per second.

donnib, 2017-08-02:
@ebaauw thanks for your answer. So in my case if i turn on a group in HASS it will tell each light to turn on with it's attributes so one call per light to deconz rest so if i have 10 lights deconz rest will get called 10times on the state then itself it will call the zigbee lights (according to you), 30 unicast zigbee messages. Does ZigBee not support one message with all attributes per light ? That would make it faster.

donnib, 2017-08-02:
> With many lights (I tested up to six), you see the lights changing independently. 
Are we talking one light per sec or is it really noticeable ?

manup, 2017-08-02:
I highly recommend to support groups for the mentioned reasons.

> Does ZigBee not support one message with all attributes per light ? That would make it faster.

Due to its generic protocol commands can't be combined. ZigBee wasn't made for high throughput. On the other side even with 150 lights group and scene commands work very well if used in moderate frequency.

ebaauw, 2017-08-02:
> if i have 10 lights deconz rest will get called 10times

Yes, PUT `/lights/1`, PUT `/lights/2`, ..., PUT `/lights/10`.

> Does ZigBee not support one message with all attributes per light ? 

No, different clusters for OnOff (0x0006), Level control (0x0008), and Color control (0x0300).  OnOff can be included with Level control, but, apparently, the Hue bridge doesn't do this.

> Are we talking one light per sec or is it really noticeable ?

It's noticeable, but I'd say all changes happen within a second.

donnib, 2017-08-02:
@manup i see, damn it :) i was hoping to not needing that. 
@ebaauw so your example where it's noticeable that is by not using groups and just control each light correct ? As far as i understand you, you don't support deConz groups right ?

ebaauw, 2017-08-02:
> I highly recommend to support groups for the mentioned reasons.

That's exactly why I supported groups very early (even before Philips has `state.any_on`).  However, I find hardly anyone is using them.  In fact, I control my lights almost exclusively from bridge/gateway rules and only update CLIPGenericFlag sensors from HomeKit.

I also found out that the Hue bridge (especially the first, round, version) was the bottleneck way before the ZigBee network was flooded.  I did some stress testing using CLIPGenericFlag sensors (so they wouldn't cause any ZigBee traffic), and found that the 1st gen bridge collapses when issuing more than 6 concurrent requests.  Consequently, I implemented throttling in homebridge-hue, with a parameter to specify the number of parallel requests, with different default values for the 1st gen and 2nd get bridge.

> so your example where it's noticeable that is by not using groups and just control each light correct ?

Yes, HomeKit group, so each light is controlled separately from Hue/deCONZ API perspective.  These tests were on a 2nd gen Hue bridge - haven't stress-tested deCONZ yet, but from what I experience, it seems more powerful than the 2nd gen Hue bridge - at least on my Rapsberry Pi 3.  Note that homebridge-hue does asynchronous I/O, i.e. it sends the second request to the API, before it has received the response to the first request.

> you don't support deConz groups right

That depends on your definition of the word "support" ;-)  I support Hue bridge groups, but haven't tested deCONZ groups yet.  I do expect these to work, now that deCONZ includes the full group resource in a GET of `/groups`.

holli73, 2017-08-02:
hello,

i use the conbee + raspbee for my house - i believe i have a big installation - i have on the conbee 60 light's and on the raspbee 5 lights - on the big env. i use only groups because with single commands you will not get what you need - f.ex in my bathroom i have 19 lights - to turn them on i have made 3 groups for different areas within the bath (so if i hit the switch only 3 group command's are send) this is the only way i could get 99.99% results.

i tried this with 19 single command's but sometimes light will not go on or off - i could only get this to work with making groups.

further i had to move from the raspbee to the conbee on a faster cpu - now i get almost realtime results - because the raspi had some latency due to the low mhz for processing and slow io - since i'm on the 3.7ghz xeon my light's just work with the conbee.

i do not use any bridge (osram/philips) only the deconz hardware.

donnib, 2017-08-03:
@holli73 that sounds interesting, what kind of lights do you have and most importantly what kind of remote/switch do you use for turning on/off the lights ? Do you see any difference in range with the raspbee vs conbee ?

holli73, 2017-08-03:
@donnib 

i use only osram lightify devices (the gu10 spots tunable white + rgbw, surface tw + rgbw led stripes and rgbw e27).

my light switches are normal 230v switch on a digitalstrom server and i use normal bash script's that involves just curl command to execute against the deconz-api or a rest-endpoint call's against automic but in the end it is just calling the deconz-api again.

i think the conbee has a better range but maybe related due that i could mount the conbee in the middle of my house under the roof (just a long usb cable needed) and the pi i could not mount that central in my house - further the housing of the raspi might influence the range as well...

ebaauw, 2017-08-03:
> i use normal bash script's that involves just curl command to execute against the deconz-api

That's a rather crude method for home automation.

If you're calling curl in the foreground, you're doing synchronous I/O to the deCONZ gateway, i.e. you send the next request only after the response of the previous request has been received.  That would seriously limit your throughput and would explain the need to use faster hardware to compensate.

If you're calling curl in the background (`curl ... &`), you'd be doing asynchronous I/O to the deCONZ gateway, i.e. you send the next request before the response of the previous request is received.  In this case, you'd be swamping the gateway sending 19 requests at the same time, causing some requests to get lost and some lights not to respond at all.

A "decent" home automation app, would do asynchronous I/O, but throttle the rate at which it sends requests.

holli73, 2017-08-03:
@ebaauw 

you are right - but if i push the switch i do not want to wait 19x250ms to have all lights on = 5 seconds or even more - so if you enter the bathroom and hit the switch 19 spots must be ON.

with 3 group commands all is on instantly f.ex.

on the pi - it took apx. 200ms to start curl - before even the request is send - so even with the 3 call's i was at 600ms just for starting curl and no light's triggered yet... (on the same box as the raspbee is running)

on the xeon box it takes 10ms to invoke the curl => 30ms and when i release the button the light is on - once my 60 lights were on the pi - i was not sure did i hit the switch? i took up to 1seconds till the lights triggered even with tcp-nodelay switch  on curl command.

i can only speak about my env. and i believe there is no way in my opinion that you can trigger 19 lights individual for a single room in a fashioned time at once - and it did look more less if you turn on the lights of airport runaway.

with using groups all lights are on at the same time

manup, 2017-08-03:
> If you're calling curl in the background (curl ... &), you'd be doing asynchronous I/O to the deCONZ gateway, i.e. you send the next request before the response of the previous request is received.

For the gateway it's not a difference between curl as background- `&` or foreground process. The received request looks the same  and is treated the same. What matters most is how many requests are you sending in parallel and frequency.

Also important is the reuse of TCP connection for consequent requests is good (Node.js and Browsers). I guess that curl won't use keep-alive since the the process is killed after handling the request — but that's totally fine in hollis use case. It would only be problematic if curl would issue heaps of requests like every 5 ms.

ebaauw, 2017-08-03:
> on the pi - it took apx. 200ms to start curl - before even the request is send - so even with the 3 call's i was at 600ms just for starting curl and no light's triggered yet... (on the same box as the raspbee is running)

Thanks, @holli73, very insightful.  I'm seeing 62ms on my Raspberry Pi 3 with a class 10 uSD card:
```
$ time curl -s -X PUT -H "Content-Type: application/json" -d '{"on":true}' "http://127.0.0.1/api/philipshue/lights/1/state"
[{"success":{"/lights/1/state/on":true}}]
real	0m0.062s
user	0m0.040s
sys	0m0.010s
```

> there is no way in my opinion that you can trigger 19 lights individual for a single room in a fashioned time at once

No there isn't - sorry if I gave the impression that there would be.  I was slightly worried that you needed to move to a faster machine.  I'm running deCONZ and homebridge with homebridge-hue on a Raspberry Pi 3 with 40 lights, 12 motions sensors, 7 switches, over a dozen CLIP sensors and ~150 gateway rules.  I don't fancy moving that to a faster machine, at least not until deCONZ would once again run on macOS.

> For the gateway it's not a difference between curl as background- & or foreground process.

No, @manup, of course not.  But executing curl in foreground from a shell script limits the number of requests in parallel to one.

>Also important is the reuse of TCP connection for consequent requests is good (Node.js and Browsers). 

Ah good one.  I tried to use keep-alive with homebridge-hue, but the Hue bridge doesn't support it, see https://developers.meethue.com/content/keep-alive.  I'll double check that homebridge-hue will set it for deCONZ.  And I'll update my mental list of deCONZ benefits over the Hue bridge...

manup, 2017-08-03:
> Ah good one. I tried to use keep-alive with homebridge-hue, but the Hue bridge doesn't support it, see https://developers.meethue.com/content/keep-alive. I'll double check that homebridge-hue will set it for deCONZ. And I'll update my mental list of deCONZ benefits over the Hue bridge...

With latest commit this should not be necessarily, for authenticated requests deCONZ will keep the connection alive for 60 s. I'll add more intelligence, so that based and *apikey* deCONZ will learn if the key is used with keep-alive in mind and act appropriate.

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/4669702ab0036ab89bf74a43de98311d5c263d34

holli73, 2017-08-03:
@ebaauw,

just some values from my current env. i still use the pi2 and pi3 -> back than my raspbee was a pi2 and if i run the curl there i have:

```
pi2:
real	0m0.105s
user	0m0.060s
sys	0m0.020s
```
so this was not fast enough for me on my pi3 now it is a bit better but not as good as on my intel but still it is just 150ms only for starting the curl command...

```
pi3:
real	0m0.077s
user	0m0.050s
sys	0m0.020s
```

```
xeon 3,7ghz
real	0m0.004s
user	0m0.000s
sys	0m0.000s
```

so it is even less than 30ms actual - curl is started in 4ms...

closed 2017-08-04 by donnib