dresden-elektronik/deconz-rest-plugin #75: Emulate events ?

donnib, 2017-08-02:
While developing is there a way to emulate and event, let's say i have a IKEA remote connected to deConz and it's shown, can i somehow make deConz send an event on the websocket as if i clicked the remote ? It would be nice if i could so i don't have to have 10 remotes on the table and find the right one to click.

ebaauw, 2017-08-02:
I don't think you can update the state of a ZHASwitch sensor from the API, but you could create a CLIPSwitch sensor for testing.  I get an event when updating `state.buttonevent` from the API:
```
Wed Aug 02 2017 09:59:20: /sensors/2/state: {"buttonevent":1002,"lastupdated":"2017-08-02T07:59:20"}
```
with the following:
```
$ ph_put /sensors/2/state '{"buttonevent": 1002}'
$ ph_get /sensors/2
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "fbfe4198e0f1522fb07052bfbefe9e2d",
  "manufacturername": "Test",
  "mode": 2,
  "modelid": "Test",
  "name": "Test",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2017-08-02T07:59:20"
  },
  "swversion": "1.0",
  "type": "CLIPSwitch",
  "uniqueid": "test"
}
```

However, when putting the same value twice, I don't receive a second event, even though `state.lastupdated` is updated.

donnib, 2017-08-02:
Ahh i see, but sadly with the same value twice that it's not issuing an event, maybe it will get fixed.

manup, 2017-08-02:
The following commit should fix the problem https://github.com/dresden-elektronik/deconz-rest-plugin/commit/c7c0489cf1556890765656f18ff4fc57040a6833
Will be available in version 2.04.60.

ebaauw, 2017-08-02:
Brilliant, and, once again, my thanks!

When is 2.04.60 due?  I'm eager to test the Xiaomi Aqara door sensor.

manup, 2017-08-02:
It's online now, hope the sensor works, couldn't test it yet.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.60-qt5.deb

ebaauw, 2017-08-02:
Sensor doesn't yet work in full (see issue #57), but v2.04.60 does issue an event each time I PUT a buttonevent, also with the same value.

EspenWathne, 2017-08-03:
@donnib , I think the best solution for doing this kind of developing us for you to loan techniques from test driven development and mock out the deconz websocket. 

ie write your own websocket server. This will allow you to be able to say to your 'mock' websocket server when to send an event and what parameters it should send. you are then in controle of creating/testing all events that the websocket can provide.

donnib, 2017-08-03:
@EspenWathne yes i know how that works and that's exactly how it should be done but in my case where this is a hobby project i don't have time to do that.

closed 2017-08-03 by donnib