dresden-elektronik/deconz-rest-plugin #76: Wrong websockets events and state when adding lights to group

donnib, 2017-08-03:
Hi,
In this particular case i had 3 lights in a group so i removed the 3 and added another 2 and here is the events on the websocket :

```
{"e":"changed","id":"2","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"39760","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"3","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"2","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":false},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":true},"t":"event"}
{"e":"changed","id":"5","r":"lights","state":{"on":true},"t":"event"}
{"e":"changed","id":"4","r":"lights","state":{"on":false},"t":"event"}
{"e":"changed","id":"3","r":"groups","state":{"any_on":false},"t":"event"}
{"e":"changed","id":"6","r":"lights","state":{"on":true},"t":"event"}
```
This goes bananas because the web thinks the lamps on and so does REST api but they are off so something goes clearly wrong. I am not sure there is a reason for sending an event for lights when joining a group ?

manup, 2017-08-03:
When lights are no more powered (I recon you mean this by *off*?) deCONZ needs a while to discover that they are not reachable anymore. Events therefore will be fired for lights until related *unreachable* state is reached.

donnib, 2017-08-03:
No, i was not clear. I don't touch the mains power on the lamps at all. All i do is i take a group and edit the lamps part of it. All lights were off while i was doing the edit of the group. When i click done (i remove 3 lights and add other 2 lights) the above events show up. The events are not correct because after the edit of the group is done the REST API reports the ligths are on which they are not.

manup, 2017-08-03:
Ah ok, that's different :) I'll check it out this really should not happen.

closed 2019-03-30 by stale[bot]