dresden-elektronik/deconz-rest-plugin #77: Changing name through API does not work for sensors

wvuyk, 2017-08-03:
If I rename a group through the REST api the result is that the command is responded to as ïnvalid command". The Webapp does also not reflect the change.
On the documentation of the API is also an error in the part on updated sensors

_PUT /api/<apikey>/rules/<id>/_
of course I guess that "rules" should be "sensors",but also the closing "/" does not follow the rest of the documentation?

Thanks,

Wim

manup, 2017-08-03:
> If I rename a group through the REST api the result is that the command is responded to as ïnvalid command". The Webapp does also not reflect the change.

Can you please provide an example how you do the request:
* URL
* content body
* how the request is made (Node.js, python, ...)
* deCONZ version used

Renaming lights, groups and sensors is testet everyday, so it should work normally.
>On the documentation of the API is also an error in the part on updated sensors

> PUT /api//rules//
of course I guess that "rules" should be "sensors",but also the closing "/" does not follow the rest of the documentation?

Good catch, will be fixed :)

wvuyk, 2017-08-03:
Version was updated today: 2.0.4.60.
URL was [serverIP]/api/[generated userkey]/sensors/2
content was {"name":"newname"]
Request is made with dot.net

I am updating my existing Philips Hue plugin to support the deCONZ implementation. So most of it is already existing?

manup, 2017-08-03:
>  content was {"name":"newname"]

Then this would be the error and should be  `{"name":"newname"}` (note curly braces at start and end).

wvuyk, 2017-08-03:
Sorry, the last brace was a typo on my end, trust me as it is the already existing reslult of the hue plugin here.

manup, 2017-08-03:
Strange, can you provide the complete error response body with HTTP status code?
There is no `invalid command` error in the sensors API.

ebaauw, 2017-08-03:
Hey Wim, nice to see you here :-)

I've updated plenty of sensor names in deCONZ doing a PUT to /api/username/sensors/# with body `{"name":"newname"}`.  I do find that deCONZ tends to return different error messages than the Hue bridge, sometimes even confusing.

wvuyk, 2017-08-03:
manup,

first of all sorry.... I had to copy the sensor definitions becuase of the differences that existed between REST API and the philips API. I made an error which resulted in sending an empty "config":[] in the command. This resulted in the invalid command. Maybe this was thrown by the Newtonsoft json interpretation. After correction, this is solved.

@Eric: Nice to see you here too! I saw your comments here, you have been busy already :-) I am also in the process of adjusting my plugin to support both worlds. Having the webhook available is for me the major reason. Different error messages for me is not really a direct issue - as long I know about it. Hope this will improve in time :-)

manup, 2017-08-03:
Cool glad that it works :) I'll close the issue for now.

closed 2017-08-03 by manup