dresden-elektronik/deconz-rest-plugin #78: scanning through REST api won't return Philips motion sensors

wvuyk, 2017-08-03:
If I inititat a scan for new devices with the API, I find new lights. But the Motion is not found, unles I do the same through the Webapp. Is this not possible?

Thanks,

Wim

manup, 2017-08-03:
The Philips hue motion sensor will only be discovered in the current beta releases.
For that:
* the network must be opened  `Settings > Open Network`
* then reset the sensor (small hole in the back)

**Note** `Settings > Scan for Devices` will not find the sensor, this function discovers only lights.

Also note the sensor is not visible in the web app but only in the REST API.

wvuyk, 2017-08-03:
manup,
The sensor is visible in the web api. I am also recieving the webhook signals perfectly? I am on the 2.04.60 build here. 
What I was trying to achieve is to discover the new device through the REST API, but I understand now this is not available yet. No problem here though, just making sure

manup, 2017-08-03:
Ah I see, well it's possible, the web app also just uses the REST API.

`POST /api/<apikey>/sensors` with an empty body will start the sensor search (same as Philips API).
I have to admit documentation lacks here and will be fixed.

The same can be achieved with Configuration API
http://dresden-elektronik.github.io/deconz-rest-doc/configuration/#modifyconfiguration

Use the `permitjoin` attribute to open the ZigBee network, this is used by the web app.

**Note** You can see what the web app is sending to the gateway in the Browser developer tools (F12 for Firefox and Chrome).

ebaauw, 2017-08-03:
There's some subtle differences with the Hue API here, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/28.

wvuyk, 2017-08-03:
Manup,

thanks! that clears it up for me. Will work on processing this. Am learning here, like eric says, there are some differences. Will catch them slowly

Thanks!

Wim

wvuyk, 2017-08-04:
Manup,

I realised today that the webapp itself is indeed not showing the motion sensors and  in de deCONS application the sensor is shown as a temperature sensor...
That is for the future I suppose?

ebaauw, 2017-08-04:
The deCONZ GUI shows ZigBee nodes.  The Hue motion sensor is one node.  If you click on the almost invisible circle at the right end of the node, the ZigBee endpoints and clusters are shown.  On the ZHA endpoint, you'll see different clusters for Temperature Measurement, Illuminance Measurement and Occupancy Detection (hope I got the names right).  The REST API turns these into separate resources for temperature, light level, and presence.   The `uniqueid` of the resource matches the ZigBee mac-address, followed by the endpoint, followed by the cluster (that's how I match the three resources into one HomeKit accessory).

For some reason beyond my comprehension, the GUI likes to show the name of the temperature resource on the node.  I've had some success changing this in `~/.local/shared/dresden elektronik/deCONZ/session.default` (make sure deCONZ is shutdown before editing this json file).

The new web app (beta out soon?) will support the motion sensors, we're promised, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/20#issuecomment-317843692.

wvuyk, 2017-08-05:
@ebaauw,

Found it. I am using the same method in Homeseer to keep them together. Also with luminaires, whitch I do not yet know for sure with deCONZ if this is going to be the same. 
But I ment to point to the webapp interface. It is showing taps and dimmer switches, but not yet the motion sensor.

wvuyk, 2017-09-13:
Gues this can be closed, looking at the new webapp

closed 2017-09-13 by wvuyk