dresden-elektronik/deconz-rest-plugin #79: Philips Hue dimmer switch does not send updates throug api or webhook

wvuyk, 2017-08-03:
I am on 2.04.60. I have added a Philips Hue Dimmer, and the device is send with the API at its initial state. But after sending this initial state, no updates are send when buttons are pressed. Not through the Webhook and not through the API? Is this something to be added still maybe?

Thanks,

Wim

manup, 2017-08-03:
Can you provide firmware version: `deCONZ > Help > About deCONZ`

deCONZ doesn't have webhooks, but there are events fired through websockets

Heres a screenshot of my Philips Hue Dimmer firing websocket events in Google Chrome.

[image]

wvuyk, 2017-08-03:
bit new in this area :-) but websockets it is. 
The firmware version in the raspbee deCONZ is 0x26110500. 
I added the DimmerSwitch on deCONZ 2.04.52, do I have to remove and re-add it maybe?

Thanks,

Wim

manup, 2017-08-03:
>bit new in this area :-) but websockets it is.
The firmware version in the raspbee deCONZ is 0x26110500.

All good :) 0x26110500 is latest version so that's fine.

>I added the DimmerSwitch on deCONZ 2.04.52, do I have to remove and re-add it maybe?

I would suggest yes. There were some improvements in the last versions which should help to get the dimmer and events setup properly.

wvuyk, 2017-08-04:
Removed the dimmer switch and re-added it. All is good now, statusses for long-press, short-press and hold are updated nicely now. One of the functions I was hoping for with the websocket :-)

Looking back I added the dimmer switch at the release how the raspbee was delivered to me last week. Think it was 2.04.35 or 36.

I am pretty happy how things work here for my plugin with the REST API and the websocket. You all are doing a good job!
Thanks,

Wim

manup, 2017-08-04:
Since it seems to work in latest releases I close the issue for now.

closed 2017-08-04 by manup