dresden-elektronik/deconz-rest-plugin #80: adding a new light dows not send a webhook "new" on every occasion

wvuyk, 2017-08-03:
Still on 2.04.60. I have tested in the past two days with adding lights. I do get a response "e":"new" once in a while, but mostly this webhook does not come, but instead a "e":"changed" most of the time.

Thanks,

Wim

ebaauw, 2017-08-03:
The new and deleted events seem only partly implemented at the moment.  I mostly see them for ZigBee sensors, but not for CLIP sensors.  The state change events seem reliable to me, although I'm not yet using them in homebridge-hue.

wvuyk, 2017-08-03:
I am here using the changed events a lot, especially for sensors. The "old" verion has to use polling to get the updates for especially sensors. In Homeseer, where my plugin works users combine all sorts of devices and want to be able to respond  to signals fast. 
Think about a dimmer switch press during evening to switch on both Zigbee and Zwave lights orclosing curtains etc. Not time critical but you can combine those actions easily. And then is waiting for polling reason to critisise its funtion.

closed 2019-03-30 by stale[bot]