dresden-elektronik/deconz-rest-plugin #81: Changing bri for a light and setting ct to same value as it is doesn't work

donnib, 2017-08-03:
Hi,
If i have a light on and i issue

`{"on": true, "bri": 194, "ct": 250}`

then i issue 

`{"on": true, "bri": 50, "ct": 250}`

the light flashes but does not use the new brightness althoug the REST API reports the set bri value. The REST API does not say that above is not allowed.

manup, 2017-08-03:
The request looks valid and should work, is this an IKEA light?

donnib, 2017-08-03:
Yet is is, i've tried this both on a E14 and a GU10 light, same thing happens

donnib, 2017-08-03:
Funny enough this (the other way around, keep `bri` fixed but change `ct` sort of works :
`{"bri": 10, "ct": 150}`
then 
`{"bri": 10, "ct": 500}`
but the lamp still flashes but at least it takes the command and uses it.

manup, 2017-08-03:
I've checked with your values `{"on": true, "bri": 194, "ct": 250}` and `{"on": true, "bri": 50, "ct": 250}`.

Both times commands were send as expected (checked in sniffer).
The light flashes, not completely dark, and turns back to bri 127.

It looks like IKEA doesn't like transitiontime.
The following works here: `{ "on": true, "bri": 50, "ct": 250, "transitiontime": 0}`

donnib, 2017-08-03:
So what does this mean ? I have to use transitiontime and set it to 0 every time i make updates ?

manup, 2017-08-03:
This seems to be a bug in the IKEA lights. If `transitiontime` is not provided in the command, it will be set to 4 by deCONZ.

donnib, 2017-08-03:
Ok i understand, funny enough there are various cases where they do support it, just not in this case for some reason.

ebaauw, 2017-08-03:
Transitiontime blues...

There is a bug in an older firmware version of the Osram lights that they won't turn off unless you specify transitiontime 0.  The Hue bridge adds a transitiontime of 4 by default as well.  This was one of the reasons I started whitelisting known lights in homebridge-hue...

closed 2019-03-30 by stale[bot]